MIT-SPARK/Kimera-Semantics #1: undefined reference to `cv::Mat::updateContinuityFlag()'

fengziyue, 2019-10-24:
I merged the kimera semantics into kimera-vio-ros and excute "catkin build", this error occurs

Errors     << pointcloud_recolor:make /home/parallels/catkin_ws/logs/pointcloud_recolor/build.make.012.log
CMakeFiles/pointcloud_recolor.dir/src/pointcloud_recolor.cpp.o: In function `PointcloudRecolor::recolorPointcloudWithImage(boost::shared_ptr<sensor_msgs::Image_<std::allocator<void> > const>, boost::shared_ptr<sensor_msgs::PointCloud2_<std::allocator<void> > const>)':
pointcloud_recolor.cpp:(.text._ZN17PointcloudRecolor26recolorPointcloudWithImageEN5boost10shared_ptrIKN11sensor_msgs6Image_ISaIvEEEEENS1_IKNS2_12PointCloud2_IS4_EEEE[_ZN17PointcloudRecolor26recolorPointcloudWithImageEN5boost10shared_ptrIKN11sensor_msgs6Image_ISaIvEEEEENS1_IKNS2_12PointCloud2_IS4_EEEE]+0x111): undefined reference to `cv::Mat::updateContinuityFlag()'
collect2: error: ld returned 1 exit status
make[2]: *** [/home/parallels/catkin_ws/devel/lib/pointcloud_recolor/pointcloud_recolor] Error 1
make[1]: *** [CMakeFiles/pointcloud_recolor.dir/all] Error 2
make: *** [all] Error 2
cd /home/parallels/catkin_ws/build/pointcloud_recolor; catkin build --get-env pointcloud_recolor | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -
...............................................................................
Failed     << pointcloud_recolor:make                           [ Exited with code 2 ]
Failed    <<< pointcloud_recolor                                [ 2.2 seconds ]
Abandoned <<< kimera_ros                                        [ Unrelated job failed ]
Abandoned <<< kimera_semantics                                  [ Unrelated job failed ]
Abandoned <<< voxblox_rviz_plugin                               [ Unrelated job failed ]
Abandoned <<< voxblox_ros                                       [ Unrelated job failed ]
Abandoned <<< kimera_semantics_ros                              [ Unrelated job failed ]
Finished  <<< KimeraVIO                                         [ 2.1 seconds ]
Finished  <<< voxblox                                           [ 0.5 seconds ]
[build] Summary: 40 of 46 packages succeeded.                                  
[build]   Ignored:   1 packages were skipped or are blacklisted.               
[build]   Warnings:  2 packages succeeded with warnings.                       
[build]   Abandoned: 5 packages were abandoned.                                
[build]   Failed:    1 packages failed.                                        
[build] Runtime: 17.3 seconds total.

ToniRV, 2019-10-24:
Hi @fengziyue,

Which opencv version are you using?
I would recommend cloning in your workspace the opencv3_catkin package to make sure you are using the same opencv version across your packages.

Related issue: https://answers.opencv.org/question/198488/undefined-reference-to-cvmatupdatecontinuityflag/

Hope it helps.

ToniRV, 2019-10-24:
- git:
    local-name: opencv3_catkin
    uri: https://github.com/ethz-asl/opencv3_catkin.git
    version: master

fengziyue, 2019-10-24:
I just excuted this:

`
mkdir -p ~/catkin_ws/src
cd ~/catkin_ws/
catkin init

echo 'source ~/catkin_ws/devel/setup.bash' >> ~/.bashrc

cd ~/catkin_ws/src
git clone git@github.com:MIT-SPARK/Kimera-VIO-ROS.git

wstool init 

wstool merge Kimera-VIO-ROS/install/kimera_ros_ssh.rosinstall

cd ~/catkin_ws/
catkin config --extend /opt/ros/melodic 
catkin config --cmake-args -DCMAKE_BUILD_TYPE=Release
catkin config --merge-devel

cd ~/catkin_ws/src
git clone git@github.com:MIT-SPARK/Kimera-Semantics.git

wstool merge Kimera-Semantics/kimera/install/kimera_semantics_ssh.rosinstall

wstool update

catkin build`

is it right?

fengziyue, 2019-10-24:
@ToniRV Hi

ToniRV, 2019-10-24:
That looks good to me.

I think I found the issue: 
What we are missing here is that the pointcloud_recolor package should depend on opencv3_catkin Yet it depends on opencv2!

Just to check:

Do you have a system wide installation of opencv?

What do you get when you type ‘opencv_version’ in the same terminal you called ‘catkin build’?

Can you try to source your catkin workspace and try to compile again so that the opencv3_catkin is used?

ToniRV, 2019-10-24:
Otw just change in the package.xml of pointcloud_recolor the build depend on opencv2 to opencv3_catkin.
I’m away from keyboard now, but feel free to raise a pull request on pointcloud recolor if that solves the issue. Thanks anyway!

fengziyue, 2019-10-24:
Yes, it solves the issue.

Thank you!

closed 2019-10-24 by fengziyue