MIT-SPARK/Kimera-Semantics #6: ros kinetic

wolf943134497, 2019-12-24:
Hi @ToniRV   Thanks for your great works!
My ros version is kinetic, something error when catkin_make the package. Does the package support ros kinetic?  I'm not very familiar with ROS 
thanks!

ToniRV, 2019-12-24:
Hi @wolf943134497! Thanks!  It should work as well in Kinetic.
If you send your error as well I might be able to help.
Also, I would recommend not to use `catkin_make` but `catkin build` as suggested in the Readme.

wolf943134497, 2019-12-27:
@ToniRV   Thanks for your suggestion! catkin build works!   Wonder why catkin_make can not work.package.xml and cmakelist file should change a little?   I usually use catkin_make to build the specified package.

ToniRV, 2019-12-28:
Not sure what might have gone wrong with `catkin_make` but I would strongly suggest to use `catkin build`. Thanks for reporting the issue!

closed 2019-12-28 by ToniRV