MIT-SPARK/Kimera-Semantics #7: Allow for geometric-only reconstruction

ToniRV, 2019-12-30:


ToniRV, 2020-02-18:
Done, just run: 
```bash
roslaunch kimera_semantics_ros kimera_semantics.launch play_bag:=true metric_semantic_reconstruction:=false
```

closed 2020-02-18 by ToniRV

amburkoff, 2020-02-20:
catkin build gives an error:

Errors     << kimera_semantics:make /home/amburkoff/cb_ws/logs/kimera_semantics/build.make.010.log                                                                                                         
/home/amburkoff/cb_ws/src/Kimera-Semantics/kimera_semantics/src/simulation/semantic_simulation_world.cpp: In member function ‘void kimera::SemanticSimulationWorld::generateSemanticSdfFromWorld(voxblox::FloatingPoint, voxblox::Layer<kimera::SemanticVoxel>*)’:
/home/amburkoff/cb_ws/src/Kimera-Semantics/kimera_semantics/src/simulation/semantic_simulation_world.cpp:84:57: error: ‘std::__cxx11::list<std::unique_ptr<voxblox::Object> > voxblox::SimulationWorld::objects_’ is private within this context
       for (const std::unique_ptr<vxb::Object>& object : objects_) {
                                                         ^~~~~~~~
In file included from /home/amburkoff/cb_ws/src/Kimera-Semantics/kimera_semantics/include/kimera_semantics/simulation/semantic_simulation_world.h:1:0,
                 from /home/amburkoff/cb_ws/src/Kimera-Semantics/kimera_semantics/src/simulation/semantic_simulation_world.cpp:1:
/home/amburkoff/cb_ws/src/voxblox/voxblox/include/voxblox/simulation/simulation_world.h:102:39: note: declared private here
   std::list<std::unique_ptr<Object> > objects_;
                                       ^~~~~~~~
/home/amburkoff/cb_ws/src/Kimera-Semantics/kimera_semantics/src/simulation/semantic_simulation_world.cpp:88:39: error: ‘class voxblox::Object’ has no member named ‘getType’; did you mean ‘Type’?
           selectSemanticLabel(object->getType(), &semantic_label);
                                       ^~~~~~~
                                       Type
make[2]: *** [CMakeFiles/kimera_semantics.dir/src/simulation/semantic_simulation_world.cpp.o] Error 1
make[1]: *** [CMakeFiles/kimera_semantics.dir/all] Error 2
make: *** [all] Error 2