MIT-SPARK/Kimera-Semantics #10: Compile issue with the newest Kimera-semantics

CaoYuchen, 2020-02-23:
Hi, I downloaded the newest version of Kimera-semantics, which is updated on 18th Feb 2020, when I tried to compile it, it reports bug as:

```
/home/caoyuchen/Documents/Kimera/catkin_ws_semantic2/src/Kimera-Semantics/kimera_semantics/src/simulation/semantic_simulation_world.cpp:84:57: error: ‘std::__cxx11::list<std::unique_ptr<voxblox::Object> > voxblox::SimulationWorld::objects_’ is private within this context
       for (const std::unique_ptr<vxb::Object>& object : objects_) {
                                                         ^~~~~~~~
In file included from /home/caoyuchen/Documents/Kimera/catkin_ws_semantic2/src/Kimera-Semantics/kimera_semantics/include/kimera_semantics/simulation/semantic_simulation_world.h:1:0,
                 from /home/caoyuchen/Documents/Kimera/catkin_ws_semantic2/src/Kimera-Semantics/kimera_semantics/src/simulation/semantic_simulation_world.cpp:1:
/home/caoyuchen/Documents/Kimera/catkin_ws_semantic2/src/voxblox/voxblox/include/voxblox/simulation/simulation_world.h:102:39: note: declared private here
   std::list<std::unique_ptr<Object> > objects_;
                                       ^~~~~~~~
/home/caoyuchen/Documents/Kimera/catkin_ws_semantic2/src/Kimera-Semantics/kimera_semantics/src/simulation/semantic_simulation_world.cpp:88:39: error: ‘class voxblox::Object’ has no member named ‘getType’; did you mean ‘Type’?
           selectSemanticLabel(object->getType(), &semantic_label);
                                       ^~~~~~~
                                       Type
make[2]: *** [CMakeFiles/kimera_semantics.dir/src/simulation/semantic_simulation_world.cpp.o] Error 1
make[1]: *** [CMakeFiles/kimera_semantics.dir/all] Error 2
make: *** [all] Error 2
```

It seems that getType() is not declared correctly for the vbx object. Can you give me some advice on it?

I come to update the newest version because I met another issue in the older version:
I mainly want to use Kimera-semantics to get 3D reconstructed mesh from my dataset, but the original version doesn't seem output any mesh result and always report another error when running the Kitti bag file as you recommended, the error goes like:

```
[ERROR] [1582470027.796071941, 43.789999999]: Failed to load nodelet [/cloudify] of type [depth_image_proc/point_cloud_xyzrgb] even after refreshing the cache: According to the loaded plugin descriptions the class depth_image_proc/point_cloud_xyzrgb with base class type nodelet::Nodelet does not exist. Declared types are  adding_images/adding_images camshift/camshift contour_moments/contour_moments convex_hull/convex_hull corner_harris/corner_harris discrete_fourier_transform/discrete_fourier_transform edge_detection/edge_detection face_detection/face_detection face_recognition/face_recognition fback_flow/fback_flow find_contours/find_contours general_contours/general_contours goodfeature_track/goodfeature_track hls_color_filter/hls_color_filter hough_circles/hough_circles hough_lines/hough_lines hsv_color_filter/hsv_color_filter lk_flow/lk_flow nodelet_tutorial_math/Plus opencv_apps/adding_images opencv_apps/camshift opencv_apps/contour_moments opencv_apps/convex_hull opencv_apps/corner_harris opencv_apps/discrete_fourier_transform opencv_apps/edge_detection opencv_apps/face_detection opencv_apps/face_recognition opencv_apps/fback_flow opencv_apps/find_contours opencv_apps/general_contours opencv_apps/goodfeature_track opencv_apps/hls_color_filter opencv_apps/hough_circles opencv_apps/hough_lines opencv_apps/hsv_color_filter opencv_apps/lk_flow opencv_apps/people_detect opencv_apps/phase_corr opencv_apps/pyramids opencv_apps/rgb_color_filter opencv_apps/segment_objects opencv_apps/simple_compressed_example opencv_apps/simple_example opencv_apps/simple_flow opencv_apps/smoothing opencv_apps/threshold opencv_apps/watershed_segmentation pcl/BAGReader pcl/BoundaryEstimation pcl/ConvexHull2D pcl/CropBox pcl/EuclideanClusterExtraction pcl/ExtractIndices pcl/ExtractPolygonalPrismData pcl/FPFHEstimation pcl/FPFHEstimationOMP pcl/MomentInvariantsEstimation pcl/MovingLeastSquares pcl/NodeletDEMUX pcl/NodeletMUX pcl/NormalEstimation pcl/NormalEstimationOMP pcl/NormalEstimationTBB pcl/PCDReader pcl/PCDWriter pcl/PFHEstimation pcl/PassThrough pcl/PointCloudConcatenateDataSynchronizer pcl/PointCloudConcatenateFieldsSynchronizer pcl/PrincipalCurvaturesEstimation pcl/ProjectInliers pcl/RadiusOutlierRemoval pcl/SACSegmentation pcl/SACSegmentationFromNormals pcl/SHOTEstimation pcl/SHOTEstimationOMP pcl/SegmentDifferences pcl/StatisticalOutlierRemoval pcl/VFHEstimation pcl/VoxelGrid people_detect/people_detect phase_corr/phase_corr pyramids/pyramids rgb_color_filter/rgb_color_filter segment_objects/segment_objects simple_compressed_example/simple_compressed_example simple_example/simple_example simple_flow/simple_flow smoothing/smoothing threshold/threshold watershed_segmentation/watershed_segmentation
[ERROR] [1582470027.796159681, 43.789999999]: The error before refreshing the cache was: According to the loaded plugin descriptions the class depth_image_proc/point_cloud_xyzrgb with base class type nodelet::Nodelet does not exist. Declared types are  adding_images/adding_images camshift/camshift contour_moments/contour_moments convex_hull/convex_hull corner_harris/corner_harris discrete_fourier_transform/discrete_fourier_transform edge_detection/edge_detection face_detection/face_detection face_recognition/face_recognition fback_flow/fback_flow find_contours/find_contours general_contours/general_contours goodfeature_track/goodfeature_track hls_color_filter/hls_color_filter hough_circles/hough_circles hough_lines/hough_lines hsv_color_filter/hsv_color_filter lk_flow/lk_flow nodelet_tutorial_math/Plus opencv_apps/adding_images opencv_apps/camshift opencv_apps/contour_moments opencv_apps/convex_hull opencv_apps/corner_harris opencv_apps/discrete_fourier_transform opencv_apps/edge_detection opencv_apps/face_detection opencv_apps/face_recognition opencv_apps/fback_flow opencv_apps/find_contours opencv_apps/general_contours opencv_apps/goodfeature_track opencv_apps/hls_color_filter opencv_apps/hough_circles opencv_apps/hough_lines opencv_apps/hsv_color_filter opencv_apps/lk_flow opencv_apps/people_detect opencv_apps/phase_corr opencv_apps/pyramids opencv_apps/rgb_color_filter opencv_apps/segment_objects opencv_apps/simple_compressed_example opencv_apps/simple_example opencv_apps/simple_flow opencv_apps/smoothing opencv_apps/threshold opencv_apps/watershed_segmentation pcl/BAGReader pcl/BoundaryEstimation pcl/ConvexHull2D pcl/CropBox pcl/EuclideanClusterExtraction pcl/ExtractIndices pcl/ExtractPolygonalPrismData pcl/FPFHEstimation pcl/FPFHEstimationOMP pcl/MomentInvariantsEstimation pcl/MovingLeastSquares pcl/NodeletDEMUX pcl/NodeletMUX pcl/NormalEstimation pcl/NormalEstimationOMP pcl/NormalEstimationTBB pcl/PCDReader pcl/PCDWriter pcl/PFHEstimation pcl/PassThrough pcl/PointCloudConcatenateDataSynchronizer pcl/PointCloudConcatenateFieldsSynchronizer pcl/PrincipalCurvaturesEstimation pcl/ProjectInliers pcl/RadiusOutlierRemoval pcl/SACSegmentation pcl/SACSegmentationFromNormals pcl/SHOTEstimation pcl/SHOTEstimationOMP pcl/SegmentDifferences pcl/StatisticalOutlierRemoval pcl/VFHEstimation pcl/VoxelGrid people_detect/people_detect phase_corr/phase_corr pyramids/pyramids rgb_color_filter/rgb_color_filter segment_objects/segment_objects simple_compressed_example/simple_compressed_example simple_example/simple_example simple_flow/simple_flow smoothing/smoothing threshold/threshold watershed_segmentation/watershed_segmentation
[FATAL] [1582470027.796337595, 43.789999999]: Failed to load nodelet '/cloudify` of type `depth_image_proc/point_cloud_xyzrgb` to manager `nodelet_manager'
[cloudify-3] process has died [pid 26910, exit code 255, cmd /opt/ros/melodic/lib/nodelet/nodelet load depth_image_proc/point_cloud_xyzrgb nodelet_manager -no-bond rgb/camera_info:=/tesse/left_cam/camera_info rgb/image_rect_color:=/tesse/segmentation depth_registered/image_rect:=/tesse/depth depth_registered/points:=/points __name:=cloudify __log:=/home/caoyuchen/.ros/log/644041f0-563c-11ea-82d2-54bf6495fecc/cloudify-3.log].
log file: /home/caoyuchen/.ros/log/644041f0-563c-11ea-82d2-54bf6495fecc/cloudify-3*.log
```

Can you also help me out with it?
Thank you so much!

ToniRV, 2020-02-23:
Thanks for reporting! I must have forgotten to adapt the API with the public voxblox, I would recommend checking out the previous commit before the last merge: 
‘’’
git checkout 7e985e96fa502e3fefaaff6c93bff2a543d85ae6
‘’’

CaoYuchen, 2020-02-24:
> Thanks for reporting! I must have forgotten to adapt the API with the public voxblox, I would recommend checking out the previous commit before the last merge:
> ‘’’
> git checkout [7e985e9](https://github.com/MIT-SPARK/Kimera-Semantics/commit/7e985e96fa502e3fefaaff6c93bff2a543d85ae6)
> ‘’’

CaoYuchen, 2020-02-24:
So now it can compile, but when I set run_stereo_dense = true, it fails to launch node of type [stereo_image_proc/stereo_image_proc]: stereo_image_proc, when I set run_stereo_dense = false, it fails to load nodelet [/cloudify] of type [depth_image_proc/point_cloud_xyzrgb]. Is seems both method are missing something.

CaoYuchen, 2020-02-24:
> So now it can compile, but when I set run_stereo_dense = true, it fails to launch node of type [stereo_image_proc/stereo_image_proc]: stereo_image_proc, when I set run_stereo_dense = false, it fails [...]

Oh sorry, it's my fault, I thought those two are self-defined APIs, but I just realize they are packages of ROS, the problems are solved! Thanks!

closed 2020-02-24 by CaoYuchen