MIT-SPARK/Kimera-Semantics #13: No algorithms outputs seen on rviz

julienip, 2020-03-04:
Hi,

Thanks for this impressive work. It looks very promising.

I checked out 7e985e9 so I can compile the code, however when I run the algorithms as instructed, nothing shows up in rviz. I have downloaded the bag and it reads successfully.
Checking the connections between the blocks in the nodes graph and I see that the bag inputs (segmentation, left_cam and, depth) are not fed into the kimera-semantic-node. 

Am I missing something to make it run successfully?

I am using Kinetic, ubuntu 16.04.

ToniRV, 2020-03-05:
Hi @julienip, thanks for reporting, the rosbag has outdated topic names (they were not following the ROS convention).
Can you try with this new rosbag? https://drive.google.com/file/d/1SG8cfJ6JEfY2PGXcxDPAMYzCcGBEh4Qq/view?usp=sharing

If that works for you I will change the links in the readme.

Thank you!

julienip, 2020-03-05:
It works! perfect! Super nice :dancing_men:

PR #14 (2020-03-06): Fix/update rosbag
Fix #13
closed 2020-03-05 by julienip