MIT-SPARK/Kimera-Semantics #15: voxblox::SimulationWorld::objects_’ is private within this context

resplendent-star, 2020-03-10:
Ubuntu 1804

catkin build in catkin_ws
_________________________________________________________________________________
Errors     << kimera_semantics:make /home/lzw/resplendent_code/k_s_ros1_ws/logs/kimera_semantics/build.make.000.log
/home/lzw/resplendent_code/k_s_ros1_ws/src/Kimera-Semantics/kimera_semantics/src/simulation/semantic_simulation_world.cpp: In member function ‘void kimera::SemanticSimulationWorld::generateSemanticSdfFromWorld(voxblox::FloatingPoint, voxblox::Layer<kimera::SemanticVoxel>*)’:
/home/lzw/resplendent_code/k_s_ros1_ws/src/Kimera-Semantics/kimera_semantics/src/simulation/semantic_simulation_world.cpp:84:57: error: ‘std::__cxx11::list<std::unique_ptr<voxblox::Object> > voxblox::SimulationWorld::objects_’ is private within this context
       for (const std::unique_ptr<vxb::Object>& object : objects_) {
                                                         ^~~~~~~~
In file included from /home/lzw/resplendent_code/k_s_ros1_ws/src/Kimera-Semantics/kimera_semantics/include/kimera_semantics/simulation/semantic_simulation_world.h:1:0,
                 from /home/lzw/resplendent_code/k_s_ros1_ws/src/Kimera-Semantics/kimera_semantics/src/simulation/semantic_simulation_world.cpp:1:
/home/lzw/resplendent_code/k_s_ros1_ws/src/voxblox/voxblox/include/voxblox/simulation/simulation_world.h:102:39: note: declared private here
   std::list<std::unique_ptr<Object> > objects_;
                                       ^~~~~~~~
/home/lzw/resplendent_code/k_s_ros1_ws/src/Kimera-Semantics/kimera_semantics/src/simulation/semantic_simulation_world.cpp:88:39: error: ‘class voxblox::Object’ has no member named ‘getType’; did you mean ‘Type’?
           selectSemanticLabel(object->getType(), &semantic_label);
                                       ^~~~~~~
                                       Type
make[2]: *** [CMakeFiles/kimera_semantics.dir/src/simulation/semantic_simulation_world.cpp.o] Error 1
make[2]: *** 正在等待未完成的任务....
make[1]: *** [CMakeFiles/kimera_semantics.dir/all] Error 2
make: *** [all] Error 2
cd /home/lzw/resplendent_code/k_s_ros1_ws/build/kimera_semantics; catkin build --get-env kimera_semantics | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -
..................................................................................
Failed     << kimera_semantics:make               [ Exited with code 2 ]          
Failed    <<< kimera_semantics                    [ 3.7 seconds ]                 
Abandoned <<< voxblox_ros                         [ Unrelated job failed ]        
Abandoned <<< kimera_semantics_ros                [ Unrelated job failed ]        
Finished  <<< voxblox_rviz_plugin                 [ 11.5 seconds ]                
[build] Summary: 11 of 14 packages succeeded.                                     
[build]   Ignored:   22 packages were skipped or are blacklisted.                 
[build]   Warnings:  2 packages succeeded with warnings.                          
[build]   Abandoned: 2 packages were abandoned.                                   
[build]   Failed:    1 packages failed.                                           
[build] Runtime: 28.1 seconds total.                                              
[build] Note: Workspace packages have changed, please re-source setup files to use them.
lzw@resplendent-star:~/resplendent_code/k_s_ros1_ws$ 
[image]

ToniRV, 2020-03-10:
Yes, sorry I pushed a change that requires my own fork of voxblox, I should tet rid of that soon. In the meantime, can you try the commit in this issue https://github.com/MIT-SPARK/Kimera-Semantics/issues/10 ?

FrozenSmile, 2020-03-21:
I met the same error and checked out the 7e985e9. Succeed in Compiling but some errors occured in roslaunch.
[ERROR] Failed to load nodelet [/cloudify] of type [depth_image_proc/point_cloud_xyzrgb]

ToniRV, 2020-03-21:
Hi @FrozenSmile , make sure you have `depth_image_proc` installed:
```
sudo apt-get install ros-melodic-depth-image-proc
```
Change melodic for your ROS distribution (i.e. kinetic etc)

ToniRV, 2020-03-30:
Necessary changes should be done here: https://github.com/ethz-asl/voxblox/pull/317

closed 2020-03-30 by ToniRV

resplendent-star, 2020-04-13:
> Yes, sorry I pushed a change that requires my own fork of voxblox, I should tet rid of that soon. In the meantime, can you try the commit in this issue #10 ?

hi,I am very sorry  to not  reply to you ,I am fresh man who are unfaimilar with ros,could plz tell me how to solve this problem  with more details?

ToniRV, 2020-04-13:
You will have to clone my fork of voxblox and remove the original voxblox:
'''
cd ~/carkin_ws/src
rm -r voxblox
git clone https://github.com/ToniRV/voxblox.git
git checkout feature/inherit_from_simulation_world
'''

You can also just wait for the changes to be merged in voxblox: 
https://github.com/ethz-asl/voxblox/pull/317

resplendent-star, 2020-04-14:
> You will have to clone my fork of voxblox and remove the original voxblox:
> 
> ```
> cd ~/carkin_ws/src
> rm -r voxblox
> git clone https://github.com/ToniRV/voxblox.git
> cd voxblox
> git checkout feature/inherit_from_simulation_world
> ```
> 
> You can also just wait for the changes to be merged in voxblox:
> [ethz-asl/voxblox#317](https://github.com/ethz-asl/voxblox/pull/317)

many thanks to you!!!!you are the god！