MIT-SPARK/Kimera-Semantics #17: Is it possible to ignore mesh construction and only use semantically segmented point cloud?

kmfrick, 2020-03-17:
Hello,
I'm evaluating the use of Kimera for my thesis project and in order to interoperate with other components the semantically segmented output must be a point cloud. 
Since the paper says that the 3D mesh is reconstructed after segmenting the point cloud, can the final mesh reconstruction be skipped? Is it only a question of modifying a launch file or do I need to alter the code (it would not be a problem, I'm just asking to better understand Kimera and where to focus my efforts). Is it the pointcloud_recolor node that does the segmentation or the  kimera_semantics_node node?
Thanks in advance!

ToniRV, 2020-03-17:
Hi @kmfrick, yes! as long as you feed a colored pointcloud with the semantic segmentation and a pose estimate, kimera-semantics will output a 3D geometric and semantic reconstruction of the scene.
Given a semantically labeled pointcloud, you can bypass the output of the `pointcloud_recolor` node and send instead your pointcloud directly to kimera_semantics_ros_node, which together with a pose estimate, should be enough for kimera-semantics to reconstruct the scene.
Are you perhaps planning to use semantically labeled LIDAR-based pointclouds?

kmfrick, 2020-03-17:
I probably did not phrase the question very well haha, sorry! My use case was having a non-labeled point cloud (from an in-house SLAM algorithm) and using Kimera to segment it while keeping it a point cloud and not reconstructing a mesh. Is this possible? Is Kimera-Semantics the right module?

kmfrick, 2020-03-24:
@ToniRV I was mistaken, Kimera does not take care of labeling the point cloud, right? [This module] (https://github.com/ToniRV/ROS_pointcloud_recolor) is probably what I'm looking for: I have semantically segmented images from which I am generating a point cloud and I want the points to be labeled. This seems to be the right tool for the job.

closed 2020-03-24 by kmfrick

ToniRV, 2020-03-24:
Correct!