MIT-SPARK/Kimera-Semantics #18: cannot launch pointcloud_recolor node

julienip, 2020-03-20:
Hi team, Hi Toni,

I am interested to run the pointcloud_recolor node as I would like to feed it with a point cloud from DSO "poincloud_in" and a semantic segmentation mask "image_in".

So I set the "run_stereo_dense" from the kimera_semantics.launch file to true and run it. However the terminal returns "cannot launch pointcloud_recolor node". 
I checked the ROS_PACKAGE_PATH and the pointcloud_recolor folder was missing. So I added it.

Now I have : cannot launch node of type [pointcloud_recolor/poincloud_recolor"].

My question:
How can I execute the "pointcloud_recolor" method if I create a rosbag that contains "pointcloud_in" topic?

Thank you for your guidance :)

ToniRV, 2020-03-20:
Hi @julienip,

Just to check, did you clone git@github.com:ToniRV/ROS_pointcloud_recolor.git in your catkin workspace?
```
git clone git@github.com:ToniRV/ROS_pointcloud_recolor.git
```
If that is the case, then can you run:
```
catkin build pointcloud_recolor
```
And if that works I would do:
```
catkin build kimera
```
Source the catkin workspace:
```
source ~/catkin_ws/devel/setup.bash
```
And run again the launch file.

If that doesn't work I might have changed the API accidentally?
Let me know :)

julienip, 2020-03-20:
It kind of works :)
I have "caught an uknown color". Results are not what I expected : 
- a gray 3D mesh 
- 3D Semantic point cloud has another set of color.
I have to look into it now. 
Thanks!

closed 2020-03-20 by julienip

ToniRV, 2020-03-20:
So the problem is that Kimera-Semantics needs a mapping from colors to semantic labels, checkout these csv files:
https://github.com/MIT-SPARK/Kimera-Semantics/tree/master/kimera_semantics_ros/cfg
I would add your custom one there and then link to your .csv file in the launch file:
https://github.com/MIT-SPARK/Kimera-Semantics/blob/master/kimera_semantics_ros/launch/kimera_semantics.launch

julienip, 2020-03-20:
Thanks for the hint. You saved me time.
In the future I would like to map the label IDs/color of my semseg to your recolor method.
I noticed this is also in your To Do list. ;)

julienip, 2020-03-20:
To have the correct colors when using "pointcloud_recolor", correct the following:
In poincloud_recolor.cpp:151 to poincloud_recolor.cpp:153, inverse the index:
*out_r = bgr [2] ; 
*out_r = bgr [1] ; 
*out_r = bgr [0] ; 
 to
*out_r = bgr [0] ; 
*out_r = bgr [1] ; 
*out_r = bgr [2] ;

ToniRV, 2020-03-29:
Thanks @julienip, I updated pointcloud_recolor.