MIT-SPARK/Kimera-Semantics #20: roslaunch error

FrozenSmile, 2020-03-21:
Thanks for your great work！ 
My first compiling failed because of the same error in issue #10 and #15，I checked out  7e985e9 and   to compile but some errors still occurred when I roslaunch.

I just change the "bag file" to the path of rosbag.  
I execute `roslaunch kimera_semantics_ros kimera_semantics.launch play_bag:=true` and the rosbag played properly but some errors ouccrred as this:

`[ERROR] Failed to load nodelet [/cloudify] of type [depth_image_proc/point_cloud_xyzrgb] even after refreshing the cache.`

`[cloudify-4] process has died [pid 123603,exit code 255, cmd /opt/ros/kinetic/lib/nodelet load depth_image_proc/point_cloud_xyzrgb noodelet_manager -no-bond rgb/camera_info:=/tesse/left_cam/camera_info rgb/image_rect_color:=/tesse/segmentation/image_raw depth_registered/image_rect:=/teese/depth/image_raw depth...`

'[image]'

'[image]'

[image]

Nothing but a moving line in my rviz. I put the error picture to show you.It looks like the same error in #10 and #13  , but I don't understand how they fixed it. By the way，I use the new rosbag you updated in the usage, so the answer in #13 didn't help me .

I guess it's the problem about node name or pkg name so I modify the launch file as the last merge, but the error still exists. 

I'm a green hand in SLAM and lack for experience, please give me some suggestion to fix errors if you can, thank you.

ToniRV, 2020-03-21:
Hi @FrozenSmile,

Thanks for the report!
As noted in #15 , try:
```
sudo apt-get install ros-melodic-depth-image-proc
```
Change melodic for your ROS distribution (i.e. kinetic etc)

FrozenSmile, 2020-03-21:
> sudo apt-get install ros-melodic-depth-image-proc
Thanks a lot for your answer ! It works immediately!

closed 2020-03-21 by FrozenSmile