MIT-SPARK/Kimera-Semantics #21: Error in installation

YJonmo, 2020-03-28:
I get error when installing this toolbox. I tried on Ubuntu 16 and 18 but both of them have error on this line: git clone git@github.com:MIT-SPARK/Kimera-Semantics.git

XX:~/catkin_ws/src$ git clone git@github.com:MIT-SPARK/Kimera-Semantics.git
Cloning into 'Kimera-Semantics'...
The authenticity of host 'github.com (13.236.229.21)' can't be established.
RSA key fingerprint is SHA256:nThbg6kXUpJWGl7E1IGOCspRomTxdCARLviKw6E5SY8.
Are you sure you want to continue connecting (yes/no)? yes
Warning: Permanently added 'github.com,13.236.229.21' (RSA) to the list of known hosts.
Permission denied (publickey).
fatal: Could not read from remote repository.

Please make sure you have the correct access rights
and the repository exists.

Any idea?

ToniRV, 2020-03-28:
Have you tried with https? 
```
https://github.com/MIT-SPARK/Kimera-Semantics.git
```
If this works, then you might need to setup your ssh keys in git: https://help.github.com/en/github/authenticating-to-github/generating-a-new-ssh-key-and-adding-it-to-the-ssh-agent

YJonmo, 2020-03-28:
Thanks Toni, it worked.

closed 2020-03-28 by ToniRV