MIT-SPARK/Kimera-Semantics #22: Installation and Demo

YJonmo, 2020-03-29:
Hi Toni,

Sorry for bothering you again.

On both Ubuntu 16 and 18 I cannot get the demo running. I did all the steps in the installation. 
But I get below image which seems there is an error in installation and demo. I put the images at the end.
Just a tiny hint, I guess in the readme this line should change

Download the demo rosbag (click here to download) and save it in: ./kimera_semantics_ros/rosbag/kimera_semantics_demo.bag

to

Download the demo rosbag (click here to download) and save it in: ./kimera_semantics_ros/rosbags/kimera_semantics_demo.bag

Installation errors:

[image]

Demo:

[image]

ToniRV, 2020-03-29:
Hi @YJonmo , please check https://github.com/MIT-SPARK/Kimera-Semantics/issues/15
I'm still working on this.

ToniRV, 2020-03-30:
Ok, so I have done the necessary changes in Voxblox, now it should be a matter of waiting for the PR to be merged (or use my fork).

https://github.com/ethz-asl/voxblox/pull/317

closed 2020-03-30 by ToniRV