MIT-SPARK/Kimera-Semantics #26: Hi @kmfrick, yes! as long as you feed a colored pointcloud with the semantic segmentation and a pose estimate, kimera-semantics will output a 3D geometric and semantic reconstruction of the scene.

julienip, 2020-04-16:
Hi Toni,

I provided a colored pointcloud to the kimera-senantics node but it doesn't output a 3D Geometric scene. 
I can see the ROS INFO : updating mesh in my terminal but not the timings messages that happen when I do have a 3D geometric scene in rviz.

You mention a pose estimate that need to be given too but I don't see that input in the kimera_semantics_node.

Thanks for your help.

Hi @kmfrick, yes! as long as you feed a colored pointcloud with the semantic segmentation and a pose estimate, kimera-semantics will output a 3D geometric and semantic reconstruction of the scene.

_Originally posted by @ToniRV in https://github.com/MIT-SPARK/Kimera-Semantics/issues/17#issuecomment-600236679_

julienip, 2020-04-16:
**After puting some traces to see the difference between when it's working and when it is not... And after commenting the following test in trasformer.cc:**
// Previous behavior was just to use the latest transform if the time is in
// the future. Now we will just wait.
  if (!tf_listener_.canTransform(to_frame, from_frame_modified,time_to_lookup)) {
      return false;
  }
 
**I have :**

[ERROR] [1587074514.738952279, 1585211039.058771156]: Error getting TF transform from sensor data: "left_cam" passed to lookupTransform argument source_frame does not exist. 

**So, I guess I have some issues with my rosbag inputs.**

in the launch file, I see :

  <arg name="sensor_frame" default="left_cam"/>

**I think I am getting closer to my answer... :)**

ToniRV, 2020-04-17:
Hi @julienip,
The pose is inferred from the TF tree in ROS (it will look for the tfs you mentioned above, `left_cam` or `left_cam_base_link` depending on what you use).
Nevertheless, you can also subscribe to a pose topic `transform`, which will be used in case you set the parameter `use_tf_transforms`  to false:
Note that these are parameters already available in voxblox, and therefore the documentation remains the same, note here https://voxblox.readthedocs.io/en/latest/pages/The-Voxblox-Node.html
```
use_tf_transforms
If true the ros tf tree will be used to get the pose of the sensor relative to the world (sensor_frame and world_frame will be used). If false the pose must be given via the transform topic.
```

nbansal90, 2021-07-23:
Hey  @ToniRV @julienip,

I am encountering a similar issue. Where everything till the point of generation of semantic pointcloud is working fine, But 3D geometric reconstruction is not happening and is thus now visualized on rviz. I think I am on the same boat, in terms that, I have ground truth pose (4 x 4) matrix available with me, (which would provide us with relative movement between consecutive frames), and I had saved it in my ROSBAG  file.

Now as suggested by @ToniRV , I have set the parameter ```use_tf_transforms``` to false before the reconstruction step in the launch file and subscribed to ```transform``` as described in the file:
```
<?xml version="1.0" encoding="ISO-8859-15"?>
<launch>
  <arg name="voxel_size"     default="0.05"/>
  <arg name="max_ray_length_m" default="5"/>
  <arg name="should_use_sim_time" default="true" />
  <param name="use_sim_time" value="$(arg should_use_sim_time)" />

  <!-- Change sensor frame to:
   - 1. VIO's estimated base_link: `left_cam_base_link`
   - 2. Or, if you want to use simulator's ground-truth: `left_cam`
  -->
  <arg name="sensor_frame" default="left_cam"/>

  <!-- If you want to play directly from a rosbag -->
  <arg name="play_bag" default="false"/>
  <!-- Let's go at twice real-time to show the speed! It can go faster, but then rviz struggles. -->
  <arg name="rosbag_rate" default="1.0"/>
  <arg name="bag_file" default="$(find kimera_semantics_ros)/rosbags/scannet_rosbag.bag"/>

  <!-- If you just want to run 3D reconstruction without semantics, set this flag to false-->
  <arg name="metric_semantic_reconstruction" default="true"/>
  <arg name="semantic_label_2_color_csv_filepath" default="$(find kimera_semantics_ros)/cfg/tesse_multiscene_office1_segmentation_mapping.csv"/>

  <!-- Input -->
  <arg name="semantic_pointcloud"         default="/semantic_pointcloud"/>
  <arg name="left_cam_info_topic"         default="/tesse/left_cam/camera_info"/>
  <arg name="right_cam_info_topic"        default="/tesse/right_cam/camera_info"/>
  <arg name="left_cam_topic"              default="/tesse/left_cam/image_raw"/>
  <arg name="right_cam_topic"             default="/tesse/right_cam/image_raw"/>
  <arg name="left_cam_segmentation_topic" default="/tesse/segmentation/image_raw"/>
  <arg name="left_cam_depth_topic"        default="/tesse/depth/image_raw"/>
  <arg name="transform"                   default="/transform"/>
  <arg name="use_freespace_pointcloud"    default="false" />
  <arg name="freespace_pointcloud"        default="/dev/null"/>

  <!-- Run rosbag if requested with play_bag arg -->
  <node name="player" pkg="rosbag" type="play" output="screen"
    args="--clock --rate $(arg rosbag_rate) $(arg bag_file)"  if="$(arg play_bag)">
    <!-- The rosbag we first generated did not follow ROS naming standards for image topics,
         so we remap the topics accordingly here.-->
    <remap from="/rgb_image_raw"     to="$(arg left_cam_topic)"/>
    <remap from="/depth_image_raw"   to="$(arg left_cam_depth_topic)"/>
    <remap from="/labels_image_raw"  to="$(arg left_cam_segmentation_topic)"/>
    <remap from="/orb_slam_pose"     to="$(arg transform)"/>
  </node>

  <!-- Generate input pointcloud with semantic labels for kimera-semantics:
     - 1. Using the depth image and registered semantic image (run_stereo_dense=false).
     - 2. Using stereo depth reconstruction (run_stereo_dense=true). -->
  <arg name="publish_point_clouds" default="true"/>
  <arg name="run_stereo_dense"     default="false"/>
  <group if="$(arg publish_point_clouds)">
    <!-- Launch Nodelet manager: used by depth_image_proc and disparity_image_proc -->
    <node pkg="nodelet" type="nodelet" name="nodelet_manager" args="manager"
      output="screen"/>
    <!-- Run stereo_dense_reconstruction_node (only if we don't use the depth image). -->
    <arg name="left_cam_stereo_depth_topic" value="/depth_image"/>
    <group if="$(arg run_stereo_dense)">
      <arg name="pointcloud" value="/points2"/>
      <include file="$(find kimera_semantics_ros)/launch/stereo_depth.launch">
<arg name="left_cam_info_topic"         value="$(arg left_cam_info_topic)"/>
        <arg name="right_cam_info_topic"        value="$(arg right_cam_info_topic)"/>
        <arg name="left_cam_topic"              value="$(arg left_cam_topic)"/>
        <arg name="right_cam_topic"             value="$(arg right_cam_topic)"/>
[... 31 lines cut ...]
    args="-alsologtostderr -colorlogtostderr" clear_params="true">
    <!-- Input -->
    <remap from="pointcloud"                to="$(arg semantic_pointcloud)"/>
    <remap from="orb_slam_pose"             to="$(arg transform)" />
    <!-- Params -->
    <param name="tsdf_voxel_size"           value="$(arg voxel_size)" />
    <param name="tsdf_voxels_per_side"      value="32" />
    <param name="max_ray_length_m"          value="$(arg max_ray_length_m)" />
    <param name="min_time_between_msgs_sec" value="0.2" />
    <param name="voxel_carving_enabled"     value="true" />
    <param name="color_mode"                value="lambert_color"/>
    <param name="use_const_weight"          value="false" />
    <param name="use_freespace_pointcloud"  value="$(arg use_freespace_pointcloud)" />
    <remap from="freespace_pointcloud"      to="$(arg freespace_pointcloud)"/>

    <param name="sensor_frame"              value="$(arg sensor_frame)"/>
    <param name="use_tf_transforms"         value="false" />

    <param name="enable_icp"                value="false" />
    <param name="icp_iterations"            value="10" />

    <param name="verbose"                   value="true" />
```

I see the mesh generation happening but 3D reconstruction is not shown on rviz. Any pointers on this front will really helpful

nbansal90, 2021-07-23:
> Hey @ToniRV @julienip,
> 
> I am encountering a similar issue. Where everything till the point of generation of semantic pointcloud is working fine, But 3D geometric reconstruction is not happening and is thus now visualized on rviz. I think I am on the same boat, in terms that, I have ground truth pose (4 x 4) matrix available with me, (which would provide us with relative movement between consecutive frames), and I had saved it in my ROSBAG file.(along with Rgb, depth and labels)
> 
> Now as suggested by @ToniRV , I have set the parameter `use_tf_transforms` to false before the reconstruction step in the launch file and subscribed to `transform`. I changed the existing `kimera_semantics.launch` file. to add my topics namely `rgb_image_raw` ,` depth_image_raw` ,` labels_image_raw` and ` orb_slam_pose`
> 
> ```
> <?xml version="1.0" encoding="ISO-8859-15"?>
> <launch>
>   <arg name="voxel_size"     default="0.05"/>
>   <arg name="max_ray_length_m" default="5"/>
>   <arg name="should_use_sim_time" default="true" />
>   <param name="use_sim_time" value="$(arg should_use_sim_time)" />
> 
>   
>   <arg name="sensor_frame" default="left_cam"/>
> 
>   
>   <arg name="play_bag" default="false"/>
>   
>   <arg name="rosbag_rate" default="1.0"/>
>   <arg name="bag_file" default="$(find kimera_semantics_ros)/rosbags/scannet_rosbag.bag"/>
> 
>   
>   <arg name="metric_semantic_reconstruction" default="true"/>
>   <arg name="semantic_label_2_color_csv_filepath" default="$(find kimera_semantics_ros)/cfg/tesse_multiscene_office1_segmentation_mapping.csv"/>
> 
>   
>   <arg name="semantic_pointcloud"         default="/semantic_pointcloud"/>
>   <arg name="left_cam_info_topic"         default="/tesse/left_cam/camera_info"/>
>   <arg name="right_cam_info_topic"        default="/tesse/right_cam/camera_info"/>
>   <arg name="left_cam_topic"              default="/tesse/left_cam/image_raw"/>
>   <arg name="right_cam_topic"             default="/tesse/right_cam/image_raw"/>
>   <arg name="left_cam_segmentation_topic" default="/tesse/segmentation/image_raw"/>
>   <arg name="left_cam_depth_topic"        default="/tesse/depth/image_raw"/>
>   <arg name="transform"                   default="/transform"/>
>   <arg name="use_freespace_pointcloud"    default="false" />
>   <arg name="freespace_pointcloud"        default="/dev/null"/>
> 
>   
>   <node name="player" pkg="rosbag" type="play" output="screen"
>     args="--clock --rate $(arg rosbag_rate) $(arg bag_file)"  if="$(arg play_bag)">
>     
>     <remap from="/rgb_image_raw"     to="$(arg left_cam_topic)"/>
>     <remap from="/depth_image_raw"   to="$(arg left_cam_depth_topic)"/>
>     <remap from="/labels_image_raw"  to="$(arg left_cam_segmentation_topic)"/>
>     <remap from="/orb_slam_pose"     to="$(arg transform)"/>
>   </node>
> 
>   
>   <arg name="publish_point_clouds" default="true"/>
>   <arg name="run_stereo_dense"     default="false"/>
>   <group if="$(arg publish_point_clouds)">
>     
>     <node pkg="nodelet" type="nodelet" name="nodelet_manager" args="manager"
>       output="screen"/>
>     
>     <arg name="left_cam_stereo_depth_topic" value="/depth_image"/>
>     <group if="$(arg run_stereo_dense)">
>       <arg name="pointcloud" value="/points2"/>
>       <include file="$(find kimera_semantics_ros)/launch/stereo_depth.launch">
> <arg name="left_cam_info_topic"         value="$(arg left_cam_info_topic)"/>
>         <arg name="right_cam_info_topic"        value="$(arg right_cam_info_topic)"/>
>         <arg name="left_cam_topic"              value="$(arg left_cam_topic)"/>
>         <arg name="right_cam_topic"             value="$(arg right_cam_topic)"/>
>         <arg name="pointcloud"                  value="$(arg pointcloud)"/>
>         <arg name="disparity_image"             value="/disparity"/>
>         <arg name="depth_image"                 value="$(arg left_cam_stereo_depth_topic)"/>
>         <arg name="convert_disparity_img_to_depth_img" value="true"/>
>       </include>
[... 27 lines cut ...]
>     <remap from="pointcloud"                to="$(arg semantic_pointcloud)"/>
>     <remap from="orb_slam_pose"             to="$(arg transform)" />
>     
>     <param name="tsdf_voxel_size"           value="$(arg voxel_size)" />
>     <param name="tsdf_voxels_per_side"      value="32" />
>     <param name="max_ray_length_m"          value="$(arg max_ray_length_m)" />
>     <param name="min_time_between_msgs_sec" value="0.2" />
>     <param name="voxel_carving_enabled"     value="true" />
>     <param name="color_mode"                value="lambert_color"/>
>     <param name="use_const_weight"          value="false" />
>     <param name="use_freespace_pointcloud"  value="$(arg use_freespace_pointcloud)" />
>     <remap from="freespace_pointcloud"      to="$(arg freespace_pointcloud)"/>
> 
>     <param name="sensor_frame"              value="$(arg sensor_frame)"/>
>     <param name="use_tf_transforms"         value="false" />
> 
>     <param name="enable_icp"                value="false" />
>     <param name="icp_iterations"            value="10" />
> 
>     <param name="verbose"                   value="true" />
> ```
> 
> I see the mesh generation happening but 3D reconstruction is not shown on rviz. Any pointers on this front will really helpful

Issue is resolved!