MIT-SPARK/Kimera-Semantics #31: No mesh published using own rosbags.

alandszhang, 2020-06-25:
Hi, @ToniRV,

Thanks for your impressive work. However, there exists a error showed as the followings when I used my own rosbag generated by RGB-D camera. In the meantime, there seems no mesh published by the nodes. Do you have any idea about the reasons. Outputs of the terminal shoe as follows.

dszhang@dszhang-hasee:~$ roslaunch kimera_semantics_ros kimera_semantics_mynteye.launch play_bag:=true metric_semantic_reconstruction:=false
... logging to /home/dszhang/.ros/log/ea3a07da-b6dd-11ea-82b1-b46921c40905/roslaunch-dszhang-hasee-2946.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://dszhang-hasee:37799/

SUMMARY
========

CLEAR PARAMETERS
 * /kimera_semantics_node/

PARAMETERS
 * /cloudify/queue_size: 20
 * /kimera_semantics_node/color_mode: lambert_color
 * /kimera_semantics_node/dynamic_semantic_labels: [20]
 * /kimera_semantics_node/enable_icp: False
 * /kimera_semantics_node/icp_iterations: 10
 * /kimera_semantics_node/max_ray_length_m: 5
 * /kimera_semantics_node/mesh_filename: /home/dszhang/cat...
 * /kimera_semantics_node/method: fast
 * /kimera_semantics_node/min_time_between_msgs_sec: 0.2
 * /kimera_semantics_node/publish_pointclouds: False
 * /kimera_semantics_node/semantic_color_mode: semantic
 * /kimera_semantics_node/semantic_label_2_color_csv_filepath: /home/dszhang/cat...
 * /kimera_semantics_node/semantic_measurement_probability: 0.8
 * /kimera_semantics_node/sensor_frame: left_cam
 * /kimera_semantics_node/slice_level: 1.0
 * /kimera_semantics_node/tsdf_voxel_size: 0.05
 * /kimera_semantics_node/tsdf_voxels_per_side: 16
 * /kimera_semantics_node/update_mesh_every_n_sec: 0.1
 * /kimera_semantics_node/use_const_weight: False
 * /kimera_semantics_node/use_freespace_pointcloud: False
 * /kimera_semantics_node/use_tf_transforms: True
 * /kimera_semantics_node/verbose: True
 * /kimera_semantics_node/voxel_carving_enabled: True
 * /rosdistro: melodic
 * /rosversion: 1.14.6
 * /use_sim_time: True

NODES
  /
    cloudify (nodelet/nodelet)
    kimera_semantics_node (voxblox_ros/tsdf_server)
    nodelet_manager (nodelet/nodelet)
    player (rosbag/play)

ROS_MASTER_URI=http://localhost:11311

process[player-1]: started with pid [2962]
process[nodelet_manager-2]: started with pid [2963]
process[cloudify-3]: started with pid [2964]
process[kimera_semantics_node-4]: started with pid [2965]
[ INFO] [1593087486.269896728]: Loading nodelet /cloudify of type depth_image_proc/point_cloud_xyzrgb to manager nodelet_manager with the following remappings:
[ INFO] [1593087486.270531303]: /depth_registered/image_rect -> /tesse/depth/image_raw
[ INFO] [1593087486.270544880]: /depth_registered/points -> /semantic_pointcloud
[ INFO] [1593087486.270551788]: /rgb/camera_info -> /tesse/left_cam/camera_info
[ INFO] [1593087486.270559571]: /rgb/image_rect_color -> /tesse/left_cam/image_raw
[ INFO] [1593087486.271091712]: Opening /home/dszhang/catkin_ws/src/Kimera-Semantics/kimera_semantics_ros/rosbags/mynteye_modified.bag
[ INFO] [1593087486.272007155]: waitForService: Service [/nodelet_manager/load_nodelet] has not been advertised, waiting...
[ INFO] [1593087486.286348887]: Initializing nodelet with 12 worker threads.
[ INFO] [1593087486.292622767]: waitForService: Service [/nodelet_manager/load_nodelet] is now available.

Waiting 0.2 seconds after advertising topics... done.
[... 82 lines cut ...]
[ INFO] [1593087490.602578056, 1593060809.207532492]: Updating mesh.937               
[ INFO] [1593087490.652391899, 1593060809.309034428]: Updating mesh.937               
[ INFO] [1593087490.702964499, 1593060809.410039801]: Updating mesh.937               
[ INFO] [1593087490.757584190, 1593060809.511100271]: Updating mesh.937               
[ INFO] [1593087490.804900776, 1593060809.612122339]: Updating mesh.937               
[ INFO] [1593087490.856610628, 1593060809.715885391]: Updating mesh.937               
[ INFO] [1593087490.906782039, 1593060809.816775843]: Updating mesh.937               
[ INFO] [1593087490.957927126, 1593060809.918178888]: Updating mesh.937               
[ INFO] [1593087491.007764375, 1593060810.018969492]: Updating mesh.937               
[ INFO] [1593087491.058997337, 1593060810.120423004]: Updating mesh.937               
[ INFO] [1593087491.101919564, 1593060810.206928112]: Updating mesh.937               
[ INFO] [1593087491.152544505, 1593060810.308278381]: Updating mesh.937               
[ INFO] [1593087491.202941581, 1593060810.409178518]: Updating mesh.937               
[ INFO] [1593087491.254175023, 1593060810.510256911]: Updating mesh.937               
[ INFO] [1593087491.305043848, 1593060810.611862486]: Updating mesh.937               
[ INFO] [1593087491.356220756, 1593060810.715454731]: Updating mesh.937               
[ INFO] [1593087491.406406952, 1593060810.816251738]: Updating mesh.937               
[ INFO] [1593087491.457295693, 1593060810.918274020]: Updating mesh.937               
[ INFO] [1593087491.509262330, 1593060811.022197446]: Updating mesh.937               
[ INFO] [1593087491.550396098, 1593060811.103635652]: Updating mesh.937               
[ INFO] [1593087491.601988169, 1593060811.206334244]: Updating mesh.937               
 [RUNNING]  Bag Time: 1593060811.293223   Duration: 9.890178 / 9.896937               
Done.
[player-1] process has finished cleanly
log file: /home/dszhang/.ros/log/ea3a07da-b6dd-11ea-82b1-b46921c40905/player-1*.log

alandszhang, 2020-06-26:
Problem solved. Thanks.

closed 2020-06-26 by alandszhang

name8997, 2020-06-30:
i have same issue, what is the solution?

Zhangppppp, 2022-03-28:
i have same issue,How did you solve it?Thanks.

prmehta24, 2023-06-25:
> 

Hi @alandszhang , I am having a similar issue using an a rosbag generated by an Intel Realsense camera. I have rgb and depth topics, and I see messages mentioning the mesh is being updated, but the mesh array remains empty. How did you solve your issue?