MIT-SPARK/Kimera-Semantics #37: catkin build error

gaotao19890725, 2020-08-20:
gaotao@gaotao-desktop:~/data/catkin_ws$ catkin_make
Base path: /home/gaotao/data/catkin_ws
Source space: /home/gaotao/data/catkin_ws/src
Build space: /home/gaotao/data/catkin_ws/build
Devel space: /home/gaotao/data/catkin_ws/devel
Install space: /home/gaotao/data/catkin_ws/install
####
#### Running command: "make cmake_check_build_system" in "/home/gaotao/data/catkin_ws/build"
####
-- Using CATKIN_DEVEL_PREFIX: /home/gaotao/data/catkin_ws/devel
-- Using CMAKE_PREFIX_PATH: /home/gaotao/data/catkin_ws/devel;/opt/ros/melodic
-- This workspace overlays: /home/gaotao/data/catkin_ws/devel;/opt/ros/melodic
-- Found PythonInterp: /usr/bin/python2 (found suitable version "2.7.17", minimum required is "2") 
-- Using PYTHON_EXECUTABLE: /usr/bin/python2
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/gaotao/data/catkin_ws/build/test_results
-- Found gtest sources under '/usr/src/googletest': gtests will be built
-- Found gmock sources under '/usr/src/googletest': gmock will be built
-- Found PythonInterp: /usr/bin/python2 (found version "2.7.17") 
-- Using Python nosetests: /usr/bin/nosetests-2.7
-- catkin 0.7.28
-- BUILD_SHARED_LIBS is on
-- BUILD_SHARED_LIBS is on
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- ~~  traversing 4 packages in topological order:
-- ~~  - catkin_simple
-- ~~  - kimera (metapackage)
WARNING: The CMakeLists.txt of the metapackage 'kimera' contains non standard content. Use the content of the following file instead: /home/gaotao/data/catkin_ws/build/catkin_generated/metapackages/kimera/CMakeLists.txt
-- ~~  - kimera_semantics
-- ~~  - kimera_semantics_ros
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
CMake Error at /opt/ros/melodic/share/catkin/cmake/catkin_workspace.cmake:100 (message):
  This workspace contains non-catkin packages in it, and catkin cannot build
  a non-homogeneous workspace without isolation.  Try the
  'catkin_make_isolated' command instead.
Call Stack (most recent call first):
  CMakeLists.txt:69 (catkin_workspace)

-- Configuring incomplete, errors occurred!
See also "/home/gaotao/data/catkin_ws/build/CMakeFiles/CMakeOutput.log".
See also "/home/gaotao/data/catkin_ws/build/CMakeFiles/CMakeError.log".
Makefile:320: recipe for target 'cmake_check_build_system' failed
make: *** [cmake_check_build_system] Error 1
Invoking "make cmake_check_build_system" failed
gaotao@gaotao-desktop:~/data/catkin_ws$ catkin_make_isolated
Base path: /home/gaotao/data/catkin_ws
Source space: /home/gaotao/data/catkin_ws/src
Build space: /home/gaotao/data/catkin_ws/build_isolated
Devel space: /home/gaotao/data/catkin_ws/devel_isolated
Install space: /home/gaotao/data/catkin_ws/install_isolated
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
~~  traversing 4 packages in topological order:
~~  - catkin_simple
~~  - kimera
~~  - kimera_semantics
~~  - kimera_semantics_ros
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

==> Processing catkin package: 'catkin_simple'
Makefile exists, skipping explicit cmake invocation...
==> make cmake_check_build_system in '/home/gaotao/data/catkin_ws/build_isolated/catkin_simple'
==> make -j4 -l4 in '/home/gaotao/data/catkin_ws/build_isolated/catkin_simple'
<== Finished processing package [1 of 4]: 'catkin_simple'

==> Processing catkin package: 'kimera'
==> Building with env: '/home/gaotao/data/catkin_ws/devel_isolated/catkin_simple/env.sh'
Makefile exists, skipping explicit cmake invocation...
==> make cmake_check_build_system in '/home/gaotao/data/catkin_ws/build_isolated/kimera'
==> make -j4 -l4 in '/home/gaotao/data/catkin_ws/build_isolated/kimera'
<== Finished processing package [2 of 4]: 'kimera'

==> Processing catkin package: 'kimera_semantics'
==> Building with env: '/home/gaotao/data/catkin_ws/devel_isolated/kimera/env.sh'
Makefile exists, skipping explicit cmake invocation...
==> make cmake_check_build_system in '/home/gaotao/data/catkin_ws/build_isolated/kimera_semantics'
==> make -j4 -l4 in '/home/gaotao/data/catkin_ws/build_isolated/kimera_semantics'
[ 37%] Building CXX object CMakeFiles/kimera_semantics.dir/src/semantic_tsdf_integrator_factory.cpp.o
[ 50%] Building CXX object CMakeFiles/kimera_semantics.dir/src/semantic_integrator_base.cpp.o
[ 50%] Building CXX object CMakeFiles/kimera_semantics.dir/src/color.cpp.o
[ 50%] Building CXX object CMakeFiles/kimera_semantics.dir/src/semantic_tsdf_integrator_fast.cpp.o
In file included from /home/gaotao/data/catkin_ws/src/Kimera-Semantics/kimera_semantics/src/semantic_tsdf_integrator_factory.cpp:36:0:
/home/gaotao/data/catkin_ws/src/Kimera-Semantics/kimera_semantics/include/kimera_semantics/semantic_tsdf_integrator_factory.h:42:10: fatal error: voxblox/integrator/tsdf_integrator.h: No such file or directory
 #include <voxblox/integrator/tsdf_integrator.h>
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
CMakeFiles/kimera_semantics.dir/build.make:134: recipe for target 'CMakeFiles/kimera_semantics.dir/src/semantic_tsdf_integrator_factory.cpp.o' failed
make[2]: *** [CMakeFiles/kimera_semantics.dir/src/semantic_tsdf_integrator_factory.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
In file included from /home/gaotao/data/catkin_ws/src/Kimera-Semantics/kimera_semantics/include/kimera_semantics/color.h:17:0,
                 from /home/gaotao/data/catkin_ws/src/Kimera-Semantics/kimera_semantics/src/color.cpp:13:
/home/gaotao/data/catkin_ws/src/Kimera-Semantics/kimera_semantics/include/kimera_semantics/common.h:9:10: fatal error: voxblox/core/block_hash.h: No such file or directory
 #include <voxblox/core/block_hash.h>
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
CMakeFiles/kimera_semantics.dir/build.make:62: recipe for target 'CMakeFiles/kimera_semantics.dir/src/color.cpp.o' failed
make[2]: *** [CMakeFiles/kimera_semantics.dir/src/color.cpp.o] Error 1
In file included from /home/gaotao/data/catkin_ws/src/Kimera-Semantics/kimera_semantics/src/semantic_tsdf_integrator_fast.cpp:37:0:
/home/gaotao/data/catkin_ws/src/Kimera-Semantics/kimera_semantics/include/kimera_semantics/semantic_tsdf_integrator_fast.h:42:10: fatal error: voxblox/integrator/tsdf_integrator.h: No such file or directory
 #include <voxblox/integrator/tsdf_integrator.h>
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/gaotao/data/catkin_ws/src/Kimera-Semantics/kimera_semantics/src/semantic_integrator_base.cpp:37:0:
/home/gaotao/data/catkin_ws/src/Kimera-Semantics/kimera_semantics/include/kimera_semantics/semantic_integrator_base.h:45:10: fatal error: voxblox/integrator/tsdf_integrator.h: No such file or directory
 #include <voxblox/integrator/tsdf_integrator.h>
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
compilation terminated.
CMakeFiles/kimera_semantics.dir/build.make:110: recipe for target 'CMakeFiles/kimera_semantics.dir/src/semantic_integrator_base.cpp.o' failed
make[2]: *** [CMakeFiles/kimera_semantics.dir/src/semantic_integrator_base.cpp.o] Error 1
CMakeFiles/kimera_semantics.dir/build.make:158: recipe for target 'CMakeFiles/kimera_semantics.dir/src/semantic_tsdf_integrator_fast.cpp.o' failed
make[2]: *** [CMakeFiles/kimera_semantics.dir/src/semantic_tsdf_integrator_fast.cpp.o] Error 1
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/kimera_semantics.dir/all' failed
make[1]: *** [CMakeFiles/kimera_semantics.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
<== Failed to process package 'kimera_semantics': 
  Command '['/home/gaotao/data/catkin_ws/devel_isolated/kimera/env.sh', 'make', '-j4', '-l4']' returned non-zero exit status 2

Reproduce this error by running:
==> cd /home/gaotao/data/catkin_ws/build_isolated/kimera_semantics && /home/gaotao/data/catkin_ws/devel_isolated/kimera/env.sh make -j4 -l4

Command failed, exiting.
gaotao@gaotao-desktop:~/data/catkin_ws$

ToniRV, 2020-10-24:
@gaotao19890725 it seems like you missed installing kimera's dependencies no?
Just run :

```
# For ssh:
wstool merge Kimera-Semantics/kimera/install/kimera_semantics_ssh.rosinstall
# For https:
#wstool merge Kimera-Semantics/kimera/install/kimera_semantics_https.rosinstall

# Download and update all dependencies
wstool update
```
in https://github.com/MIT-SPARK/Kimera-Semantics#b-kimera-semantics-installation

closed 2020-10-24 by ToniRV

shaoxh, 2021-10-15:
Hello!
I used wstool update failed because of connection time out, so I clone every project as .rosinstall name it.
The "voxblox" part is also cloned in this path: /home/tongji/catkin_ws_kimera/src.
But this error is still there.
A lot of  code files in kimera_semantic could not find those "*.h" files  in "voxblox".
Could you please tell the reason for that?  @ToniRV @gaotao19890725 
Thanks!