MIT-SPARK/Kimera-Semantics #38: Pointcloud queue getting too long error

miajl, 2020-09-10:
I am currently trying to run VIO and Semantics using ROS and the Intel 435i and I am having trouble getting semantics to pair up nicely.

Currently doing the following in separate terminals

```
roscore
roslaunch realsense2_camera rs_camera.launch unite_imu_method:=linear_interpolation enable_infra1:=true enable_infra2:=true enable_gyro:=true enable_accel:=true
roslaunch kimera_vio_ros kimera_vio_ros_realsense_IR.launch
roslaunch kimera_semantics_ros kimera_metric_realsense.launch register_depth:=true
```

I get semantics to start running and it seems like it is working but after about 10-15 seconds I get this error and new points stop being added to the mesh.

`[ERROR] [1599771723.658597680]: Input pointcloud queue getting too long! Dropping some pointclouds. Either unable to look up transform timestamps or the processing is taking too long.`

Here is my ros log from a test

```
[rosmaster.main][INFO] 2020-09-10 17:01:17,460: initialization complete, waiting for shutdown
[rosmaster.main][INFO] 2020-09-10 17:01:17,460: Starting ROS Master Node
[xmlrpc][INFO] 2020-09-10 17:01:17,460: XML-RPC server binding to 0.0.0.0:11311
[xmlrpc][INFO] 2020-09-10 17:01:17,461: Started XML-RPC server [http://L4MLAROCCALinux:11311/]
[rosmaster.master][INFO] 2020-09-10 17:01:17,461: Master initialized: port[11311], uri[http://L4MLAROCCALinux:11311/]
[xmlrpc][INFO] 2020-09-10 17:01:17,461: xml rpc node: starting XML-RPC server
[rosmaster.master][INFO] 2020-09-10 17:01:17,479: +PARAM [/run_id] by /roslaunch
[rosmaster.master][INFO] 2020-09-10 17:01:17,480: +PARAM [/roslaunch/uris/host_l4mlaroccalinux__45933] by /roslaunch
[rosmaster.master][INFO] 2020-09-10 17:01:17,547: +PARAM [/rosversion] by /roslaunch
[rosmaster.master][INFO] 2020-09-10 17:01:17,547: +PARAM [/rosdistro] by /roslaunch
[rosmaster.master][INFO] 2020-09-10 17:01:17,551: +SERVICE [/rosout/get_loggers] /rosout http://L4MLAROCCALinux:38471/
[rosmaster.master][INFO] 2020-09-10 17:01:17,551: +SERVICE [/rosout/set_logger_level] /rosout http://L4MLAROCCALinux:38471/
[rosmaster.master][INFO] 2020-09-10 17:01:17,552: +PUB [/rosout_agg] /rosout http://L4MLAROCCALinux:38471/
[rosmaster.master][INFO] 2020-09-10 17:01:17,554: +SUB [/rosout] /rosout http://L4MLAROCCALinux:38471/
[rosmaster.master][INFO] 2020-09-10 17:01:22,380: +PARAM [/roslaunch/uris/host_l4mlaroccalinux__37505] by /roslaunch
[rosmaster.master][INFO] 2020-09-10 17:01:22,386: +PARAM [/camera/realsense2_camera/enable_color] by /roslaunch
[rosmaster.master][INFO] 2020-09-10 17:01:22,386: +PARAM [/camera/realsense2_camera/linear_accel_cov] by /roslaunch
[rosmaster.master][INFO] 2020-09-10 17:01:22,386: +PARAM [/camera/realsense2_camera/fisheye1_optical_frame_id] by /roslaunch
[rosmaster.master][INFO] 2020-09-10 17:01:22,386: +PARAM [/camera/realsense2_camera/enable_sync] by /roslaunch
[rosmaster.master][INFO] 2020-09-10 17:01:22,386: +PARAM [/camera/realsense2_camera/pose_frame_id] by /roslaunch
[rosmaster.master][INFO] 2020-09-10 17:01:22,386: +PARAM [/camera/realsense2_camera/pointcloud_texture_index] by /roslaunch
[rosmaster.master][INFO] 2020-09-10 17:01:22,387: +PARAM [/camera/realsense2_camera/enable_fisheye] by /roslaunch
[rosmaster.master][INFO] 2020-09-10 17:01:22,387: +PARAM [/camera/realsense2_camera/tf_publish_rate] by /roslaunch
[rosmaster.master][INFO] 2020-09-10 17:01:22,387: +PARAM [/camera/realsense2_camera/topic_odom_in] by /roslaunch
[rosmaster.master][INFO] 2020-09-10 17:01:22,387: +PARAM [/camera/realsense2_camera/odom_frame_id] by /roslaunch
[rosmaster.master][INFO] 2020-09-10 17:01:22,387: +PARAM [/camera/realsense2_camera/aligned_depth_to_infra1_frame_id] by /roslaunch
[rosmaster.master][INFO] 2020-09-10 17:01:22,387: +PARAM [/camera/realsense2_camera/initial_reset] by /roslaunch
[rosmaster.master][INFO] 2020-09-10 17:01:22,387: +PARAM [/camera/realsense2_camera/accel_fps] by /roslaunch
[rosmaster.master][INFO] 2020-09-10 17:01:22,387: +PARAM [/camera/realsense2_camera/gyro_optical_frame_id] by /roslaunch
[rosmaster.master][INFO] 2020-09-10 17:01:22,387: +PARAM [/camera/realsense2_camera/color_frame_id] by /roslaunch
[rosmaster.master][INFO] 2020-09-10 17:01:22,387: +PARAM [/camera/realsense2_camera/depth_width] by /roslaunch
[rosmaster.master][INFO] 2020-09-10 17:01:22,387: +PARAM [/camera/realsense2_camera/enable_infra1] by /roslaunch
[rosmaster.master][INFO] 2020-09-10 17:01:22,387: +PARAM [/camera/realsense2_camera/color_width] by /roslaunch
[rosmaster.master][INFO] 2020-09-10 17:01:22,387: +PARAM [/camera/realsense2_camera/enable_pose] by /roslaunch
[rosmaster.master][INFO] 2020-09-10 17:01:22,387: +PARAM [/camera/realsense2_camera/enable_infra2] by /roslaunch
[rosmaster.master][INFO] 2020-09-10 17:01:22,388: +PARAM [/camera/realsense2_camera/fisheye_fps] by /roslaunch
[rosmaster.master][INFO] 2020-09-10 17:01:22,388: +PARAM [/camera/realsense2_camera/infra_fps] by /roslaunch
[rosmaster.master][INFO] 2020-09-10 17:01:22,388: +PARAM [/camera/realsense2_camera/color_optical_frame_id] by /roslaunch
[rosmaster.master][INFO] 2020-09-10 17:01:22,388: +PARAM [/camera/realsense2_camera/fisheye_width] by /roslaunch
[rosmaster.master][INFO] 2020-09-10 17:01:22,388: +PARAM [/camera/realsense2_camera/align_depth] by /roslaunch
[rosmaster.master][INFO] 2020-09-10 17:01:22,388: +PARAM [/camera/realsense2_camera/infra1_optical_frame_id] by /roslaunch
[rosmaster.master][INFO] 2020-09-10 17:01:22,388: +PARAM [/camera/realsense2_camera/infra2_optical_frame_id] by /roslaunch
[rosmaster.master][INFO] 2020-09-10 17:01:22,388: +PARAM [/camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id] by /roslaunch
[rosmaster.master][INFO] 2020-09-10 17:01:22,388: +PARAM [/camera/realsense2_camera/depth_optical_frame_id] by /roslaunch
[rosmaster.master][INFO] 2020-09-10 17:01:22,388: +PARAM [/camera/realsense2_camera/fisheye2_optical_frame_id] by /roslaunch
[rosmaster.master][INFO] 2020-09-10 17:01:22,388: +PARAM [/camera/realsense2_camera/aligned_depth_to_infra2_frame_id] by /roslaunch
[rosmaster.master][INFO] 2020-09-10 17:01:22,388: +PARAM [/camera/realsense2_camera/fisheye_height] by /roslaunch
[rosmaster.master][INFO] 2020-09-10 17:01:22,388: +PARAM [/camera/realsense2_camera/enable_fisheye2] by /roslaunch
[rosmaster.master][INFO] 2020-09-10 17:01:22,388: +PARAM [/camera/realsense2_camera/enable_depth] by /roslaunch
[rosmaster.master][INFO] 2020-09-10 17:01:22,388: +PARAM [/camera/realsense2_camera/enable_fisheye1] by /roslaunch
[rosmaster.master][INFO] 2020-09-10 17:01:22,389: +PARAM [/camera/realsense2_camera/depth_fps] by /roslaunch
[rosmaster.master][INFO] 2020-09-10 17:01:22,389: +PARAM [/camera/realsense2_camera/pose_optical_frame_id] by /roslaunch
[... 787 lines cut ...]
  File "/usr/lib/python2.7/xmlrpclib.py", line 1311, in single_request
    self.send_content(h, request_body)
  File "/usr/lib/python2.7/xmlrpclib.py", line 1459, in send_content
    connection.endheaders(request_body)
  File "/usr/lib/python2.7/httplib.py", line 1078, in endheaders
    self._send_output(message_body)
  File "/usr/lib/python2.7/httplib.py", line 894, in _send_output
    self.send(msg)
  File "/usr/lib/python2.7/httplib.py", line 856, in send
    self.connect()
  File "/usr/lib/python2.7/httplib.py", line 833, in connect
    self.timeout, self.source_address)
  File "/usr/lib/python2.7/socket.py", line 575, in create_connection
    raise err
error: [Errno 111] Connection refused

[rosmaster.master][INFO] 2020-09-10 17:02:12,699: -PUB [/rosout_agg] /rosout http://L4MLAROCCALinux:38471/
[rosmaster.master][INFO] 2020-09-10 17:02:12,699: -SUB [/rosout] /rosout http://L4MLAROCCALinux:38471/
[rosmaster.master][INFO] 2020-09-10 17:02:12,700: -SERVICE [/rosout/get_loggers] /rosout rosrpc://L4MLAROCCALinux:45957
[rosmaster.master][INFO] 2020-09-10 17:02:12,700: -SERVICE [/rosout/set_logger_level] /rosout rosrpc://L4MLAROCCALinux:45957
[rosmaster.main][INFO] 2020-09-10 17:02:12,780: keyboard interrupt, will exit
[rosmaster.main][INFO] 2020-09-10 17:02:12,780: stopping master...
[rospy.core][INFO] 2020-09-10 17:02:12,780: signal_shutdown [atexit]

```

hyclover, 2020-09-28:
Dear miajl，I want to run kimera-semantics with Realsense D435i now, could you please tell me how to make it work?  Is other components like kimera-VIO or kimera-RPGO  used when running kimera-semantics?

Abirami-Segar, 2021-11-04:
Hi Miajil,

i have the exact same issue as you have mentioned. Did you figure the solution out? pls share any useful information/solution. thank you

BADAL244, 2023-05-05:
[ERROR] [1599771723.658597680]: Input pointcloud queue getting too long! Dropping some pointclouds. Either unable to look up transform timestamps or the processing is taking too long. 

I also getting the same error.

Xuefei98, 2023-07-20:
I am also running into the same error. [ERROR] [1689876513.606770073, 60.195000000]: Input pointcloud queue getting too long! Dropping some pointclouds. Either unable to look up transform timestamps or the processing is taking too long.
Using docker and ubuntu18.04. If I run kimera-semantic on my host machine, the mesh display is normal and does not return any error.

bqm1111, 2025-03-20:
Have anyone figured out how to fix this issue?