MIT-SPARK/Kimera-Semantics #39: Error reading version line when launch kimera_semantics.launch play_bag:=true

AndrewWilesRG, 2020-09-16:
dear repository,
I installed kimera-semantics and ros  correctly following the instruction online, when I use the command : roslaunch kimera_semantics_ros kimera_semantic.launch play_bag:=true  ,the error is throwed out: Error reading version line
what's wrong  with this ?
my OS version is :ubunt16.04
ROS version: ros kinetic

ToniRV, 2020-09-16:
I can't reproduce this error, but, alternatively, just play the rosbag in another terminal (and leave play_bag as its default which is `false`). Or do you really read to use the rosbag to play in the same launch file?

ToniRV, 2020-10-24:
Closing, but feel free to re-open if you still have this error.

closed 2020-10-24 by ToniRV