MIT-SPARK/Kimera-Semantics #41: Is it possible to run Kimera Semantics without ROS?

apache8080, 2020-10-19:
As the title says. I want to be able to run Kimera Semantics without ROS just like Kimera-VIO.

ToniRV, 2020-10-19:
Hi @apache8080, unfortunately right now, kimera_semantics doesn't have a non-ROS interface (we only have the kimera_semantics_ros as an interface). It should be possible to create your own interface using the kimera_semantics library which is ROS agnostic, but we are not planning, at least on the short term, to add a non-ROS interface.

apache8080, 2020-10-19:
Thanks for the response. The one part I am confused about is where does Kimear Semantics get the pose from. The voxblox node subscirbes to transform so is KimeraVIO-ROS publishing that out?

apache8080, 2020-10-19:
Also I am guessing the main reason to use ROS is that visualization is easier.

ToniRV, 2020-10-19:
> Thanks for the response. The one part I am confused about is where does Kimear Semantics get the pose from. The voxblox node subscirbes to transform so is KimeraVIO-ROS publishing that out?
 Yes!
> Also I am guessing the main reason to use ROS is that visualization is easier.
Visualization and communication between processes.

apache8080, 2020-10-19:
Is the topic called transform in Kimera-VIO-ROS? Would you be able to point me to the spot in Kimera-VIO-ROS that publishes this data and where in Kimera-semantics this data gets subscribed?
Thanks for the help.

ToniRV, 2020-10-24:
Hi @apache8080 , I just updated the readme in #44, so that it is easier to run Kimera-Semantics + Kimera-VIO
Answering your question:
> Is the topic called transform in Kimera-VIO-ROS? Would you be able to point me to the spot in Kimera-VIO-ROS that publishes this data and where in Kimera-semantics this data gets subscribed?

Basically, Kimera-Semantics looks on the TF tree what is the transformation between `world_frame` and `sensor_frame`.
These are parameters in the launch files:
- `world_frame` is typically set to `world`
- `sensor_frame` is typically set to `cam0` in Euroc or `left_cam` in the Simulator (rosbag provided in readme).
See https://github.com/MIT-SPARK/Kimera-Semantics/blob/aeb7639b89ad531bdc7edb2798a48f1a42386f20/kimera_semantics_ros/launch/kimera_semantics_euroc.launch#L8
See https://github.com/MIT-SPARK/Kimera-Semantics/blob/aeb7639b89ad531bdc7edb2798a48f1a42386f20/kimera_semantics_ros/launch/kimera_semantics.launch#L12

Now, Kimera-VIO-ROS outputs two TF frames:
- `base_link` frame: which is the estimated pose of the IMU frame with respect to the world frame.
- A static frame published in `/tf_static` from the estimated `base_link` frame to the `cam0` frame (in case of Euroc) or `left_frame` (in simulation). Since this tf corresponds to the transformation btw IMU and left camera, it is static (or assumed to be static ofc) and it is taken from the LeftCameraParams.yaml file in Kimera-VIO's params folder OR from the rostopics (e.g. `/cam0/camera_info` in case of Euroc).

Hope this helps!

closed 2020-10-24 by ToniRV