MIT-SPARK/Kimera-Semantics #45: Caught an unknown color

p-akanksha, 2020-10-27:
Hi, 

I am trying to run Kimera Semantic on the data collected from the Tesse simulator (the one used in the [goseek-challenge](https://github.com/MIT-TESSE/goseek-challenge/blob/master/Instructions.md#installation)). I keep getting the following error: 
```
E1027 12:28:32.195211 24858 color.cpp:73] Caught an unknown color: 
RGB: 143 171 0 255
```

Also, the reconstructed point cloud does not have associated semantic labels. Here is an image of the point cloud. 
[image]

I am unable to figure out what might be causing this error. Any help is highly appreciated.

ToniRV, 2020-10-27:
Hi @p-akanksha,

There is a `.csv` file inside the [cfg](https://github.com/MIT-SPARK/Kimera-Semantics/tree/master/kimera_semantics_ros/cfg) folder (configuration) that specifies the mapping between colors and object ids.

Depending on the colors in the semantic image, you might need to specify one csv vs another.
I think the one for go-seek should be already the default one:
https://github.com/MIT-SPARK/Kimera-Semantics/blob/7198e5a3998321671ace2cbafe0f8a640f07cac0/kimera_semantics_ros/launch/kimera_semantics.launch#L126-L127

But maybe you can try replacing `office1` in the line above by `office2`?

ZacRavichandran, 2020-10-27:
Hi @p-akanksha 

Like @ToniRV mentioned, the error indicates that there was an object not specified in the configuration `.csv`. There's a configuration for the goseek-challenge scenes, [`goseek_segmentation_mapping_dense.csv`](https://github.com/MIT-TESSE/tesse-gym-bridge/blob/master/cfg/goseek/goseek_segmentation_mapping_dense.csv), that should contain all the necessary information. Perhaps you could try replacing `tesse_multiscene_office1_segmentation_mapping.csv` with that?

p-akanksha, 2020-10-28:
Thanks! Replacing ```tesse_multiscene_office1_segmentation_mapping.csv``` with ```goseek_segmentation_mapping_dense.csv``` worked perfectly. 

This has also increased the speed of 3d reconstruction. Earlier it felt like it was processing frames in batches, now it reconstructs immediately. Do you think the delay was because it was trying to look for the objects in the configuration file and unable to find them?

ToniRV, 2020-10-28:
Hi  @p-akanksha,

Glad it worked!

The lookup of the semantic id for a given color should have a constant time complexity as we are using `std::unordered_map`.
Nevertheless, if the `getSemanticLabelFromColor` function doesn't find a color, it will start printing errors with the missing colors, this is not particularly fast.

p-akanksha, 2020-10-29:
That makes sense. Thanks for the prompt response.

closed 2020-10-29 by p-akanksha