MIT-SPARK/Kimera-Semantics #46: Catkin build error: cannot find macros.h

shonigmann, 2020-12-15:
I recently pulled the latest commits and found that I could no longer build Kimera-Semantics, getting the following error:

```
Starting  >>> kimera_semantics                                                                        
Starting  >>> voxblox_rviz_plugin                                                                     
______________________________________________________________________________________________________
Errors     << kimera_semantics:cmake /home/simon/catkin_ws/logs/kimera_semantics/build.cmake.000.log  
CMake Error at /home/simon/catkin_ws/devel/share/catkin_simple/cmake/catkin_simple-extras.cmake:150 (add_library):
  Cannot find source file:

    include/kimera_semantics/macros.h

  Tried extensions .c .C .c++ .cc .cpp .cxx .cu .m .M .mm .h .hh .h++ .hm
  .hpp .hxx .in .txx
Call Stack (most recent call first):
  CMakeLists.txt:8 (cs_add_library)

CMake Error at /home/simon/catkin_ws/devel/share/catkin_simple/cmake/catkin_simple-extras.cmake:150 (add_library):
  No SOURCES given to target: kimera_semantics
Call Stack (most recent call first):
  CMakeLists.txt:8 (cs_add_library)

CMake Generate step failed.  Build files cannot be regenerated correctly.
cd /home/simon/catkin_ws/build/kimera_semantics; catkin build --get-env kimera_semantics | catkin env -si  /usr/local/bin/cmake /home/simon/catkin_ws/src/Kimera-Semantics/kimera_semantics --no-warn-unused-cli -DCATKIN_DEVEL_PREFIX=/home/simon/catkin_ws/devel -DCMAKE_INSTALL_PREFIX=/home/simon/catkin_ws/install -DCMAKE_BUILD_TYPE=Release; cd -
......................................................................................................
Failed     << kimera_semantics:cmake                            [ Exited with code 1 ]                
Failed    <<< kimera_semantics                                  [ 1.3 seconds ]                       
Abandoned <<< voxblox_ros                                       [ Unrelated job failed ]              
Abandoned <<< kimera_semantics_ros                              [ Unrelated job failed ]              
```

Things build as expected by commenting out the `include/kimera_semantics/macros.h` line from `Kimera-Semantics/kimera_semantics/CMakeLists.txt`, but I haven't tested thoroughly enough to see if that breaks anything else..

Is a file missing from the latest commit or is there an extraneous line in the CMakeLists? Or am have I made another mistake I'm not aware of?

Thanks for the help!

ToniRV, 2020-12-15:
Hi @shonigmann,

I accidentally added that line in the cmakelists.txt.
If you remove it and open a PR I'll merge it asap.

Thanks!

shonigmann, 2020-12-15:
Hi @ToniRV - done. Thanks for the quick response.

closed 2020-12-15 by shonigmann