MIT-SPARK/Kimera-Semantics #51: Choosing a neural network for semantic segmentation

amburkoff, 2021-01-19:
Can you please tell me which neural networks of segmentation do you recommend for use with kimera?  What are the requirements for it?  How to choose the right one?  I collected a real dataset from my robot and launched it on kimera, now I need to select a neural network for semantic segmentation and run semantics together with kimera.

ToniRV, 2021-02-09:
Hi @amburkoff , Mask-RCNN is the easiest/fastest to use, but any that outputs semantic labels on a per-pixel basis will do.
We are planning to release a how-to wiki page to make this process easier.

nbansal90, 2021-07-20:
> Hi @amburkoff , Mask-RCNN is the easiest/fastest to use, but any that outputs semantic labels on a per-pixel basis will do.
> We are planning to release a how-to wiki page to make this process easier.

Hey @ToniRV , Do we have the wiki ready for integrating off the shelf segmentation model (Example Mask-RCNN) with the Kimera . Please share if yes, otherwise a small (step wise) and approx write-up about the required set up would be much appreciated. Thank you