MIT-SPARK/Kimera-Semantics #55: Parametrize number of semantic labels

RozDavid, 2021-02-03:
Hey @ToniRV,

First of all, thank you very much for this work, in simulation it works amazing, but would like to have some question about how to make the transition to real world data. 

So I realized, when included a custom CNN ROS node for running the segmentation, that the number of labels in confidence matrices are fixed. 

The Struct declaration where you declare this is [here](https://github.com/MIT-SPARK/Kimera-Semantics/blob/f75d5db0c4fc33d869ab65996d210a6213c0309b/kimera_semantics/include/kimera_semantics/semantic_voxel.h#L21). 

While I was inspecting how to change this to Dynamic size matrices I realized that the static struct is used several times and in multiple places. For the first try it didn't seem to be too straightforward to fix this problem, but if you could help with a few guiding ideas where to replace I would be able to provide a pull request with the fix. 

My first guesses would be:
- load this param from the Ros parameter server to [SemanticConfig](https://github.com/RozDavid/Kimera-Semantics/blob/f75d5db0c4fc33d869ab65996d210a6213c0309b/kimera_semantics_ros/src/ros_params.cpp#L38),
-  replace the [Struct](https://github.com/MIT-SPARK/Kimera-Semantics/blob/f75d5db0c4fc33d869ab65996d210a6213c0309b/kimera_semantics/include/kimera_semantics/semantic_voxel.h#L21) to dynamic template sizes,
- resize in in the [SemanticIntegratorBase::setSemanticProbabilities()](https://github.com/MIT-SPARK/Kimera-Semantics/blob/f75d5db0c4fc33d869ab65996d210a6213c0309b/kimera_semantics/src/semantic_integrator_base.cpp#L93) 
-  also initialize in the _fast_ and _merge_ implementations. 
-  finally change the Constant initialization in multiple places with the parametrized log likelihood

Could you help what I would be missing and what could go wrong after change? 

Thanks a lot,
David

AnhTuDo1998, 2021-02-04:
@RozDavid I realized the same thing today when working on a different segmentation model trained on ADE20K. Seems like the value is defined in this [header file](https://github.com/MIT-SPARK/Kimera-Semantics/blob/master/kimera_semantics/include/kimera_semantics/common.h) for kimera_semantics module.

Something i observed for my side is when changing that constant to my model's number of label, running the kimera semantics will fail this assertion code in semantic_integrator_base.cpp. Not sure if you know why it happens, I just disabled it for now.

```
CHECK_NEAR(
      semantic_log_likelihood_.sum(),
      kTotalNumberOfLabels * log_match_probability_ +
          std::pow(kTotalNumberOfLabels, 2) * log_non_match_probability_ -
          kTotalNumberOfLabels * log_non_match_probability_,
      1000 * vxb::kFloatEpsilon);
```

RozDavid, 2021-02-04:
Hey @AnhTuDo1998,

Yeah, I started to work on implementing the dynamic number of labels yesterday, and met with the same problem when we set the total number over 25ish. It is only a rounding error I believe, so we just need to adapt the treshold with the number of labels there, or just manually set it a megnitude larger. For sanity check I think it will still stand. 

I believe the bigger challenge is here the initialization of SemanticVoxel priors when assigning new voxblox blocks with a constant matrix, that's size and value depends on the parameter. With non-fixed sized Eigen matrices we cannot use the original Structs that was defined [here](https://github.com/MIT-SPARK/Kimera-Semantics/blob/f75d5db0c4fc33d869ab65996d210a6213c0309b/kimera_semantics/include/kimera_semantics/semantic_voxel.h#L21).

ToniRV, 2021-02-04:
Hi! That sounds like a great idea! I'll get back to you later with suggestions :+1: 
The idea of fixed Eigen types was just to make it faster, but dynamic matrices should also work.

ToniRV, 2021-02-08:
I added my comments to #56, that is definitely the way to go!