MIT-SPARK/Kimera-Semantics #58: 3D reconstruction (without semantics) using Intel Realsense D435i is not working

hieronymusma, 2021-02-15:
Hi everybody,

Unfortunately I have some problems building the 3D reconstruction  using the Intel Realsense D435i.

Whenever I start Kimera-Semantics I can see the pose estimation in rviz but no reconstruction. In the Terminal I only get 
`[ INFO] [1613430674.227458481]: Updating mesh.`

I am executing the following commands:
1. Start roscore
`roscore`

2. Start camera node
`roslaunch realsense2_camera rs_camera.launch unite_imu_method:=linear_interpolation enable_infra:=true enable_infra1:=true enable_infra2:=true enable_gyro:=true enable_accel:=true align_depth:=true`

3. Disable emitter
`rosrun dynamic_reconfigure dynparam set /camera/stereo_module emitter_enabled 0`

4. Start Kimera-VIO
`roslaunch kimera_vio_ros kimera_vio_ros_realsense_IR.launch run_stereo_dense:=true`

5. Start Kimera-Semantics
`roslaunch kimera_semantics_ros kimera_metric_realsense.launch`

I've changed following parameters in the launch file of kimera semantics:

`should_use_sim_time:=false`
`run_stereo_dense:=true` # I've tested it with both false and true but neither works
`metric_semantic_reconstruction:=false`

`sensor_frame:=left_cam_base_link`

`left_cam_info_topic:=/camera/aligned_depth_to_color/camera_info`
`left_cam_topic:=/camera/color/image_raw`
`left_cam_segmentation_topic:=/camera/color/semantic_image`
`left_cam_depth_topic:=/camera/aligned_depth_to_color/image_raw`

Here is a visualization of my ROS topics:

[image]

The visualization of Kimera-VIO works fine. However I've encountered the issue, that after some time the visualization of Kimera-VIO stops when I start Kimera-Semantics but the status messages of Kimera-VIO are still printed.

I also receive sometimes the following message after startup of Kimera-VIO:
`W0215 23:15:30.628022  5205 DataProviderModule.cpp:118] Asking for data before start of IMU stream, from timestamp: 1613430930557135336 to timestamp: 1613430930626134204`. After removing and plugging in the camera again it mostly works.

I run the complete system inside a Docker container, could that be the problem?
Ubuntu 18.04
ROS Melodic 1.14.10
Realsense Firmware: 05.12.11.00

I'm unsure, if I haven't set a topic properly or forgot something else.

I would really appreciate it if someone could guide me into the right direction. I'm relatively new to ROS, therefore I don't know how to analyze the problem correctly.

Thank you very much,
Maurice

mbed92, 2021-02-16:
Same here. IThe error "Asking for data before start of IMU stream" happens always when the ```enable_accel:=true``` flag is set.

DragonOnGit, 2021-12-07:
Hey, i have got the same problem but a little bit more. I try to write a new launch file based on `kimera_semantics_euroc.launch and kimera_vio_ros_euroc.launch`. 
It can also work with vio but no mesh information in rviz.
The file cantains:
```
kimera_vio_ros_euroc.launch based
<launch>
  <arg name="dataset_name" value="RealSenseIR"/>
  <arg name="online" default="True" />

  <arg name="log_output" default="false"/>
  <arg name="log_output_path"
       default="$(find kimera_vio_ros)/output_logs/"
       if="$(arg log_output)"/>
  <arg name="log_gt_data" default="false"/>
  <arg name="gt_topic" default=""/>
  <!-- Set use_sim_time to true if you use rosbag with clock argument -->
  <arg name="should_use_sim_time" value="false"/>

  <!-- Only used when parsing a rosbag -->
  <!-- <arg name="rosbag_path" default="default.bag"
       unless="$(arg online)"/>"/> -->

  <!-- Frame IDs. These DO NOT match frame id's on the video streams, as the 
	RealSense and Kimera publish conflicting Tf's -->
  <arg name="world_frame_id"     value="world"/>
  <arg name="base_link_frame_id" value="base_link"/>
  <arg name="map_frame_id"       value="map"/>
  <arg name="left_cam_frame_id"  value="left_cam"/>
  <arg name="right_cam_frame_id" value="right_cam"/>

  <!-- Perform stereo dense reconstruction? -->
  <arg name="run_stereo_dense"     value="true"/>
  <group ns="stereo_gray" if="$(arg run_stereo_dense)">
    <!-- StereoSGBM -->
    <node name="dense_stereo" pkg="image_undistort" type="dense_stereo_node" >

      <param name="input_camera_info_from_ros_params" value="false"/>

      <param name="first_camera_namespace"  value="/camera/infra1"/>
      <param name="second_camera_namespace" value="/camera/infra2"/>
      <param name="first_camera_info"  value="/camera/infra1/camera_info"/>
      <param name="second_camera_info"  value="/camera/infra2/camera_info"/>
      <param name="first_output_frame"      value="/camera/infra1"/>
      <param name="second_output_frame"     value="/camera/infra2"/>

      <param name="depth/use_sgbm"          value="true"/>
      <param name="depth/do_median_blur"    value="false"/>
      <param name="depth/use_mode_HH"       value="true"/>

      <param name="scale"                   value="1.0"/>
      <param name="process_every_nth_frame" value="1"/>
      <param name="publish_tf"              value="false"/>

      <remap from="raw/first/image"         to="/camera/infra1/image_rect_raw"/>
      <remap from="raw/second/image"        to="/camera/infra2/image_rect_raw"/>
      <remap from="raw/first/camera_info"   to="/camera/infra1/camera_info"/>
      <remap from="raw/second/camera_info"  to="/camera/infra2/camera_info"/>

    </node>
  </group>

  <!-- Subscribed Topics -->
  <arg name="left_cam_topic"          value="/camera/infra1/image_rect_raw"/>
  <arg name="left_cam_info_topic"     value="/camera/infra1/camera_info"/>
  <arg name="right_cam_topic"         value="/camera/infra2/image_rect_raw"/>
  <arg name="right_cam_info_topic"    value="/camera/infra2/camera_info"/>
  <arg name="imu_topic"               value="/camera/imu"/>

  <!-- Launch actual pipeline -->
  <include file="$(find kimera_vio_ros)/launch/kimera_vio_ros.launch"
           pass_all_args="true"/>

 <!-- Launch static TF node from base_link to velo_link -->
 <node pkg="tf" type="static_transform_publisher" name="velo_link_broadcaster"
   args="0 0 0 0 0 0 1 base_link velo_link 100"/>

</launch>
```

and 

```
kimera_semantics_euroc.launch based
<?xml version="1.0" encoding="ISO-8859-15"?>
<launch>
  <arg name="voxel_size"     default="0.10"/>
  <arg name="max_ray_length_m" default="5"/>
  <param name="should_use_sim_time" value="false" />

  <!-- We use VIO's estimated base_link: `left_cam_base_link` -->
  <arg name="sensor_frame" default="left_cam_base_link"/>

  <!-- We just want to run metric reconstruction -->
  <arg name="metric_semantic_reconstruction" default="false"/>

  <!-- Input -->
  <arg name="semantic_pointcloud"         default="/stereo_gray/dense_stereo/pointcloud"/>
  <!-- Only used for dense stereo depth, but we do that in Kimera VIO ROS -->
  <arg name="left_cam_info_topic"         default="/camera/infra1/camera_info"/>
  <arg name="right_cam_info_topic"        default="/camera/infra2/camera_info"/>
  <arg name="left_cam_topic"              default="/camera/infra1/image_rect_raw"/>
  <arg name="right_cam_topic"             default="/camera/infra2/image_rect_raw"/>
  <!-- We don't have semantics in Euroc -->
  <arg name="left_cam_segmentation_topic" default="/dev/null"/>
  <!-- We don't use a depth topic, but a pointcloud directly -->
  <arg name="left_cam_depth_topic"        default="/dev/null"/>
  <arg name="use_freespace_pointcloud"    default="true" />
  <arg name="freespace_pointcloud"        default="/stereo_gray/dense_stereo/freespace_pointcloud"/>

  <!-- We don't need to run stereo depth or publish point clouds since it is done in Kimera VIO ROS already.
       So we just need to subscribe in "semantic_pointcloud" to the actual pointcloud from Kimera VIO ROS -->
  <arg name="publish_point_clouds" default="false"/>
  <arg name="run_stereo_dense"     default="false"/>

  <arg name="pkg_type"    default="kimera_semantics_ros"  if="$(arg metric_semantic_reconstruction)"/>
  <arg name="server_type" default="kimera_semantics_node" if="$(arg metric_semantic_reconstruction)"/>
  <arg name="pkg_type"    default="voxblox_ros" unless="$(arg metric_semantic_reconstruction)"/>
  <arg name="server_type" default="tsdf_server" unless="$(arg metric_semantic_reconstruction)"/>

  <include file="$(find kimera_semantics_ros)/launch/kimera_semantics.launch" pass_all_args="true"/>
</launch>
```

These file work fine with everything but mesh.

moksh019, 2022-01-14:
I also faced same issue.
I stored the data in a bag file and run the bag file at rate=0.1 . This worked for me.

CCCC-user, 2022-06-15:
hello,I have the same problem. Have you solved it?

prmehta24, 2023-06-24:
@moksh-401-511 I am trying to do 3D reconstruction(no semantics) with an intel realsense d435i bagfile. 

I am doing the following steps:
1. roscore.
2. rviz -d $(rospack find kimera_semantics_ros)/rviz/kimera_realsense_metric.rviz
3. roslaunch kimera_semantics_ros kimera_metric_realsense.launch play_bag:=true (I replaced the kimera_semantics_demo.bag with my bag. metric_semantic_reconstruction is default set to false. should_use_sim_time is set to true.)

However, I can not see the mesh reconstruction and have an error global status error - fixed frame - unknown frame world

[image]

moksh-401-511, 2023-06-25:
> @moksh-401-511 I am trying to do 3D reconstruction(no semantics) with an intel realsense d435i bagfile.
> 
> I am doing the following steps:
> 
> 1. roscore.
> 2. rviz -d $(rospack find kimera_semantics_ros)/rviz/kimera_realsense_metric.rviz
> 3. roslaunch kimera_semantics_ros kimera_metric_realsense.launch play_bag:=true (I replaced the kimera_semantics_demo.bag with my bag. metric_semantic_reconstruction is default set to false. should_use_sim_time is set to true.)
> 
> However, I can not see the mesh reconstruction and have an error global status error - fixed frame - unknown frame world
> 
> Do you have an idea of how to solve this issue and view the 3D reconstruction?
> 
> [image]

Can you do rosbag info of your bag file and share the screenshot?

prmehta24, 2023-06-25:
> Can you do rosbag info of your bag file and share the screenshot?

Sure, here it is.

[image]

moksh-401-511, 2023-06-26:
As you know, to generate mesh, Kimera-semantics needs to know where your camera is w.r.t. world frame. So, you need to run kimera-vio-ros as mentioned [here](https://github.com/MIT-SPARK/Kimera-Semantics#with-kimera-vio) to track your camera position in a world frame. It will require the IMU data of d435i. <br>
If you're using bag file then you might need to launch this [file](https://github.com/MIT-SPARK/Kimera-Semantics/blob/master/kimera_semantics_ros/launch/kimera_semantics_rosbag.launch). Setup the sensor-frame and other necessary parameters. Then give it a try.