MIT-SPARK/Kimera-Semantics #68: Error when launching example

DaddyWesker, 2021-06-09:
Hello.

I'm getting error when launching Euroca dataset usage. Putting this `roslaunch kimera_vio_ros kimera_vio_ros_euroc.launch run_stereo_dense:=true` to the terminal gives me 

`... logging to /home/daddywesker/.ros/log/d25bbaac-c904-11eb-bb6d-cbdd1b9e0c72/roslaunch-daddywesker-ROG-Strix-G712LW-G712LW-29718.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://daddywesker-ROG-Strix-G712LW-G712LW:33919/

SUMMARY
========

PARAMETERS
 * /gt_accel_bias: [-0.012492, 0.547...
 * /gt_gyro_bias: [-0.002229, 0.020...
 * /kimera_vio_ros/kimera_vio_ros_node/base_link_frame_id: base_link
 * /kimera_vio_ros/kimera_vio_ros_node/ground_truth_odometry_rosbag_topic: 
 * /kimera_vio_ros/kimera_vio_ros_node/imu_rosbag_topic: /imu0
 * /kimera_vio_ros/kimera_vio_ros_node/left_cam_frame_id: cam0
 * /kimera_vio_ros/kimera_vio_ros_node/left_cam_rosbag_topic: /cam0/image_raw
 * /kimera_vio_ros/kimera_vio_ros_node/log_gt_data: False
 * /kimera_vio_ros/kimera_vio_ros_node/map_frame_id: map
 * /kimera_vio_ros/kimera_vio_ros_node/mono_ransac_threshold: 30
 * /kimera_vio_ros/kimera_vio_ros_node/online_run: False
 * /kimera_vio_ros/kimera_vio_ros_node/params_folder_path: /home/daddywesker...
 * /kimera_vio_ros/kimera_vio_ros_node/position_det_threshold: 0.3
 * /kimera_vio_ros/kimera_vio_ros_node/right_cam_frame_id: cam1
 * /kimera_vio_ros/kimera_vio_ros_node/right_cam_rosbag_topic: /cam1/image_raw
 * /kimera_vio_ros/kimera_vio_ros_node/rosbag_path: 
 * /kimera_vio_ros/kimera_vio_ros_node/stereo_ransac_threshold: 20
 * /kimera_vio_ros/kimera_vio_ros_node/use_lcd: False
 * /kimera_vio_ros/kimera_vio_ros_node/use_online_cam_params: False
 * /kimera_vio_ros/kimera_vio_ros_node/velocity_det_threshold: 0.1
 * /kimera_vio_ros/kimera_vio_ros_node/world_frame_id: world
 * /kimera_vio_ros/posegraph_viewer/frame_id: world
 * /rosdistro: noetic
 * /rosversion: 1.15.11
 * /stereo_gray/dense_stereo/cam0/T_cam_imu: [[0.0140678333899...
 * /stereo_gray/dense_stereo/cam0/cam_overlaps: [1]
 * /stereo_gray/dense_stereo/cam0/camera_model: pinhole
 * /stereo_gray/dense_stereo/cam0/distortion_coeffs: [-0.0116026114041...
 * /stereo_gray/dense_stereo/cam0/distortion_model: equidistant
 * /stereo_gray/dense_stereo/cam0/flip_camera: True
 * /stereo_gray/dense_stereo/cam0/intrinsics: [461.341850302615...
 * /stereo_gray/dense_stereo/cam0/resolution: [752, 480]
 * /stereo_gray/dense_stereo/cam0/rostopic: /cam0/image_raw
 * /stereo_gray/dense_stereo/cam1/T_cam_imu: [[0.0118105423902...
 * /stereo_gray/dense_stereo/cam1/T_cn_cnm1: [[0.9999964092230...
 * /stereo_gray/dense_stereo/cam1/cam_overlaps: [0]
 * /stereo_gray/dense_stereo/cam1/camera_model: pinhole
 * /stereo_gray/dense_stereo/cam1/distortion_coeffs: [0.00573062561909...
 * /stereo_gray/dense_stereo/cam1/distortion_model: equidistant
 * /stereo_gray/dense_stereo/cam1/flip_camera: True
 * /stereo_gray/dense_stereo/cam1/intrinsics: [459.682043146586...
 * /stereo_gray/dense_stereo/cam1/resolution: [752, 480]
 * /stereo_gray/dense_stereo/cam1/rostopic: /cam1/image_raw
 * /stereo_gray/dense_stereo/depth/do_median_blur: False
 * /stereo_gray/dense_stereo/depth/use_mode_HH: True
 * /stereo_gray/dense_stereo/depth/use_sgbm: True
 * /stereo_gray/dense_stereo/first_camera_namespace: cam0
 * /stereo_gray/dense_stereo/first_output_frame: cam0
 * /stereo_gray/dense_stereo/input_camera_info_from_ros_params: True
 * /stereo_gray/dense_stereo/process_every_nth_frame: 1
 * /stereo_gray/dense_stereo/publish_tf: False
 * /stereo_gray/dense_stereo/scale: 1.0
 * /stereo_gray/dense_stereo/second_camera_namespace: cam1
 * /stereo_gray/dense_stereo/second_output_frame: cam1
 * /use_sim_time: True

NODES
  /kimera_vio_ros/
    kimera_vio_ros_node (kimera_vio_ros/kimera_vio_ros_node)
    posegraph_viewer (pose_graph_tools/visualizer_node)
  /stereo_gray/
    dense_stereo (image_undistort/dense_stereo_node)

ROS_MASTER_URI=http://localhost:11311

process[stereo_gray/dense_stereo-1]: started with pid [29744]
process[kimera_vio_ros/kimera_vio_ros_node-2]: started with pid [29745]
process[kimera_vio_ros/posegraph_viewer-3]: started with pid [29746]
[ INFO] [1623237838.797831267]: Initializing pose graph visualizer
I0609 14:23:58.988674 29745 Logger.cpp:59] Opening output file: traj_gt.csv
I0609 14:23:58.989750 29745 RosBagDataProvider.cpp:52] Constructing RosbagDataProvider from path: 
 - Rosbag Path: 
With ROS topics: 
 - Left cam: /cam0/image_raw
 - Right cam: /cam1/image_raw
 - IMU: /imu0
 - GT odom: 
F0609 14:23:58.989765 29745 RosBagDataProvider.cpp:60] Check failed: !rosbag_path_.empty() 
*** Check failure stack trace: ***
    @     0x7fd9839ca703  google::LogMessage::Fail()
    @     0x7fd9839cf4ab  google::LogMessage::SendToLog()
    @     0x7fd9839ca3ff  google::LogMessage::Flush()
    @     0x7fd9839cac2f  google::LogMessageFatal::~LogMessageFatal()
    @     0x7fd9837605e8  VIO::RosbagDataProvider::RosbagDataProvider()
    @     0x7fd98374a2bc  VIO::KimeraVioRos::createDataProvider()
    @     0x7fd98374ad4a  VIO::KimeraVioRos::runKimeraVio()
    @     0x563082034734  main
    @     0x7fd982d0f0b3  __libc_start_main
    @     0x563082034b4e  _start
[kimera_vio_ros/kimera_vio_ros_node-2] process has died [pid 29745, exit code -6, cmd /home/daddywesker/Dioram/TestingRGB/Kimera/VIO-ROS/devel/lib/kimera_vio_ros/kimera_vio_ros_node --use_lcd=false --vocabulary_path=/home/daddywesker/Dioram/TestingRGB/Kimera/VIO-ROS/src/Kimera-VIO/vocabulary/ORBvoc.yml --flagfile=/home/daddywesker/Dioram/TestingRGB/Kimera/VIO-ROS/src/Kimera-VIO/params/Euroc/flags/Mesher.flags --flagfile=/home/daddywesker/Dioram/TestingRGB/Kimera/VIO-ROS/src/Kimera-VIO/params/Euroc/flags/VioBackend.flags --flagfile=/home/daddywesker/Dioram/TestingRGB/Kimera/VIO-ROS/src/Kimera-VIO/params/Euroc/flags/RegularVioBackend.flags --flagfile=/home/daddywesker/Dioram/TestingRGB/Kimera/VIO-ROS/src/Kimera-VIO/params/Euroc/flags/Visualizer3D.flags --logtostderr=1 --colorlogtostderr=1 --log_prefix=1 --v=0 --log_output=false --log_euroc_gt_data=false --output_path=/home/daddywesker/Dioram/TestingRGB/Kimera/VIO-ROS/src/Kimera-VIO-ROS/output_logs/ --viz_type=0 --visualize=true reinit_flag:=reinit_flag reinit_pose:=reinit_pose odometry:=odometry resiliency:=resiliency imu_bias:=imu_bias optimized_trajectory:=optimized_trajectory pose_graph:=pose_graph mesh:=mesh frontend_stats:=frontend_stats debug_mesh_img/image_raw:=debug_mesh_img/image_raw feature_tracks/image_raw:=feature_tracks/image_raw time_horizon_pointcloud:=time_horizon_pointcloud __name:=kimera_vio_ros_node __log:=/home/daddywesker/.ros/log/d25bbaac-c904-11eb-bb6d-cbdd1b9e0c72/kimera_vio_ros-kimera_vio_ros_node-2.log].
log file: /home/daddywesker/.ros/log/d25bbaac-c904-11eb-bb6d-cbdd1b9e0c72/kimera_vio_ros-kimera_vio_ros_node-2*.log
`

Maybe command is not full and i need to set this rosbag_path somewhere?

DaddyWesker, 2021-06-09:
This command `roslaunch kimera_vio_ros kimera_vio_ros_euroc.launch run_stereo_dense:=true rosbag_path:=/home/daddywesker/datasets/Euroc/V1_01_easy.bag` pushed this a little bit further till the 
`I0609 14:34:35.239802 32536 RosBagDataProvider.cpp:83] Initialize Rosbag Data Provider.
I0609 14:34:35.239812 32536 RosBagDataProvider.cpp:201] Parsing rosbag data.
F0609 14:34:35.275579 32536 RosBagDataProvider.cpp:222] Check failed: vio_params_.backend_params_->autoInitialize_ == 1 (0 vs. 1) Requested ground-truth initialization, but no gt_odom_topic was given. Make sure you set ground_truth_odometry_rosbag_topic, or turn autoInitialize to false in BackendParameters.yaml.`

DaddyWesker, 2021-06-09:
I've looked at BackendParameters for Euroc and sought that autoInitialize is already 0 (False). So, i've switched it to 1 (True) and launched. It now goes further and i'm recieveing this
`E0609 15:08:23.955045 34223 Mesher.cpp:1729] Delaunay Triangle out of image (size: x: 0, y: 0, height: 480, width 752
 Triangle: x, y: 
2256, 0
0, 2256
687.563, 353.45
`

and many more of that with different coordinates. I thought it is fine and launched all other commands, but rviz looks like this and i'm thinking that this is wrong. 
[image]

DaddyWesker, 2021-06-10:
Well, that is how i made it work:

1. Open a new terminal, run: roscore
2. In another terminal, launch Kimera-VIO-ROS:
roslaunch kimera_vio_ros kimera_vio_ros_euroc.launch run_stereo_dense:=true  rosbag_path:=/path/to/rosbag
3. In another terminal, launch Kimera-Semantics:
roslaunch kimera_semantics_ros kimera_semantics_euroc.launch
4. Finally, in another terminal, run Rviz for visualization:
rviz -d $(rospack find kimera_semantics_ros)/rviz/kimera_semantics_euroc.rviz

For some reason, rosbag play V1_01_easy.bag --clock just making all the area white blank. Unfortunately, without it nothing happens inside "Left img" window, but at least it works.

[image]

closed 2021-06-10 by DaddyWesker

DaddyWesker, 2021-06-11:
Sorry, here is a more adequate solution. 
Instead of this 
`roslaunch kimera_semantics_ros kimera_semantics.launch play_bag:=true`

launch this
`roslaunch kimera_semantics_ros kimera_semantics.launch play_bag:=true online:=true`

and it will wait for rosplay command and everything will be as planned.
[image]

ChiyuanFeng, 2022-06-16:
> 
Requested ground-truth initialization, but no gt_odom_topic was given. Make sure you set ground_truth_odometry_rosbag_topic, or turn autoInitialize to false in BackendParameters.yaml.

Hello Sir, I encountered the same problem. Could you give me more info about this? I can't find  BackendParameters.yaml.

prmehta24, 2023-06-02:
@DaddyWesker Were you able to create a 3D voxelized world using Kimera-Semantics with the Euroc dataset?

Did you get an output similar to running Kimera Semantics on kimera_semantics_demo.bag, but in black and white?

I am able to get an output similar to what you have shown in the screenshot above, but I have not been able to get the 3D voxelized world for the Euroc dataset.

prmehta24, 2023-06-19:
To get a 3D voxelized world in black and white with the Euroc dataset, I followed the instructions given in the README with one change:

I replaced this line:

`roslaunch kimera_vio_ros kimera_vio_ros_euroc.launch run_stereo_dense:=true`

with:

`roslaunch kimera_vio_ros kimera_vio_ros_euroc.launch run_stereo_dense:=true online:=true`

The output once Kimera Semantics has been run for the Euroc dataset looks like this:

[image]