MIT-SPARK/Kimera-Semantics #74: Why can't I achieve the effect of the paper after rebuilding the map?

HongLouyemeng, 2021-11-16:
[image]
The above is the result of my reconstruction using "uhumans_12" data, and the following is the result in the paper:
Is it the problem of data set or parameters? Can you give me some suggestions?
[image]

ToniRV, 2021-11-16:
@HongLouyemeng the figure in the paper is from the uHumans1 dataset: http://mit.edu/sparklab/datasets/uHumans/

https://drive.google.com/file/d/1c6vAHmCd9onycQnCVnw6rK5mNPJ-dC3j/view?usp=sharing

HongLouyemeng, 2021-11-17:
> @HongLouyemeng the figure in the paper is from the uHumans1 dataset: http://mit.edu/sparklab/datasets/uHumans/
> 
> https://drive.google.com/file/d/1c6vAHmCd9onycQnCVnw6rK5mNPJ-dC3j/view?usp=sharing

Thank you for your reply

closed 2021-11-17 by HongLouyemeng