MIT-SPARK/Kimera-Semantics #76: How to run Kimera-semantics with semantic labels using kimera_semantics_demo.bag with Kimera-VIO-ROS?

aswingururaj, 2022-03-06:
Hi,
     I can run kimera-semantics with kimera_semantics_demo.bag(with semantics) and run Euroc dataset with dense stereo reconstruction(without semantics). I want to run them together to test the entire pipeline with Kimera-VIO-ROS giving the odometry estimates, Kimera semantics taking in the semantic point cloud to generate 3D metric-semantic reconstruction. Can anyone help me out on how I should set up Kimera to get the entire pipeline running?

Thanks!

closed 2022-03-12 by aswingururaj

dongho-Han, 2023-04-25:
Hello, @aswingururaj .
I'm trying to run kimera-semantics with kimera-VIO, but I have hard time running the entire pipeline in NVIDIA-docker(cudagl) system(ubuntu: 18.04, ROS: melodic). Can you share the way you solved?
Thanks!

aswingururaj, 2023-05-01:
Hi @dongho-Han! I didn't run the pipeline using the docker environment. I followed the installation and usage instructions under README to set up the pipeline. I am not sure what your error is. But if I remember correctly, I made a mistake while remapping the ROS topics, which is why I faced this issue. Once I corrected it, I was able to get the expected results.