MIT-SPARK/Kimera-Semantics #86: Kimera Semantics not generating 3D mesh with RealSense D455

shabpompeiano, 2023-09-08:
**Description:**
I am trying to obtain a 3D mesh of an entire room using Kimera Vio and Kimera Semantics in conjunction with a RealSense D455 camera. Although Kimera Semantics is designed to produce this, I am not able to see the 3D mesh. In rviz, the display shows only black (refer to attached images).

I am attaching the launch files and the output to rqt_graph too. 

**Commands:**
``` roslaunch realsense2_camera rs_camera.launch \
unite_imu_method:=linear_interpolation \
enable_accel:=true \
enable_gyro:=true \
enable_imu:=true \
enable_color:=true \
enable_depth:=true \
enable_infra1:=true \
enable_infra2:=true \
infra_width:=640 \
infra_height:=480 \
infra_fps:=15 \
depth_width:=640 \
depth_height:=480 \
depth_fps:=15 \
color_width:=1280 \
color_height:=720 \
color_fps:=15 \
initial_reset:=true 
```

`rosrun dynamic_reconfigure dynparam set /camera/stereo_module emitter_enabled 0`

`roslaunch kimera_vio_ros kimera_vio_ros_realsense_IR.launch`

`rviz -d $(rospack find kimera_vio_ros)/rviz/kimera_vio_euroc.rviz`

`roslaunch kimera_semantics_ros kimera_metric_realsense.launch register_depth:=true`

`rviz -d $(rospack find kimera_semantics_ros)/rviz/kimera_realsense_metric.rviz`

**Console output:**
```
 * /kimera_vio_ros/kimera_vio_ros_node/position_det_threshold: 0.3
 * /kimera_vio_ros/kimera_vio_ros_node/right_cam_frame_id: right_cam
 * /kimera_vio_ros/kimera_vio_ros_node/stereo_ransac_threshold: 20
 * /kimera_vio_ros/kimera_vio_ros_node/use_lcd: False
 * /kimera_vio_ros/kimera_vio_ros_node/use_online_cam_params: False
 * /kimera_vio_ros/kimera_vio_ros_node/velocity_det_threshold: 0.1
 * /kimera_vio_ros/kimera_vio_ros_node/world_frame_id: world
 * /kimera_vio_ros/posegraph_viewer/frame_id: world
 * /rosdistro: melodic
 * /rosversion: 1.14.13
 * /use_sim_time: False

NODES
  /
    velo_link_broadcaster (tf/static_transform_publisher)
  /kimera_vio_ros/
    kimera_vio_ros_node (kimera_vio_ros/kimera_vio_ros_node)
    posegraph_viewer (pose_graph_tools/visualizer_node)

ROS_MASTER_URI=http://localhost:11311

process[kimera_vio_ros/kimera_vio_ros_node-1]: started with pid [3994]
process[kimera_vio_ros/posegraph_viewer-2]: started with pid [3995]
process[velo_link_broadcaster-3]: started with pid [3996]
[ INFO] [1694179178.729510298]: Initializing pose graph visualizer
I0908 15:19:38.830406  3994 RegularVioBackend.cpp:114] Using Regular VIO Backend.
I0908 15:19:38.830544  3994 PipelineModule.h:449] MISO Pipeline Module: Display has no output queue registered.
I0908 15:19:38.830662  3994 Pipeline.cpp:296] Pipeline Modules launched (parallel_run set to 1).
I0908 15:19:38.830775  3994 KimeraVioRos.cpp:137] Statistics
-----------                                  #  Log Hz  {avg     +- std    }  [min,max]
Display [ms]                                  0 
[... 310 lines cut ...]
  <arg name="right_cam_topic" value="/camera/infra2/image_rect_raw"/>
  <arg name="imu_topic"       value="/camera/imu"/>

  <!-- Launch actual pipeline -->
  <include file="$(find kimera_vio_ros)/launch/kimera_vio_ros.launch"
           pass_all_args="true"/>

 <!-- Launch static TF node from base_link to velo_link -->
 <node pkg="tf" type="static_transform_publisher" name="velo_link_broadcaster"
   args="0 0 0 0 0 0 1 base_link velo_link 100"/>

</launch>

```
[image]
[image]
[image]

Please give also the following information:
* Kimera-Semantics branch, tag or commit used: commit 1ed69c61f4d1d71a812336574ef298e6210ebcd2
* GTSAM version used:
* OpenCV version used: type `opencv_version`
* Operating system and version (e.g. Ubuntu 16.04 or Windows 10): Ubuntu 18.04
* ROS Distribution (kinetic, melodic, …): melodic
* Did you change the source code? (yes / no): no (only launch files)

jolags313, 2023-09-29:
@shabpompeiano Have you solved this issue? I'm experiencing the same thing. 

Also, for me, nothing is streaming on the /camera/aligned_depth_to_color/camera_info or /camera/aligned_depth_to_color/image_raw topics. Could you tell me how you got the depth image in the second screenshot you have?

shabpompeiano, 2023-10-02:
If you set the argument `align_depth:=true` when initializing realsense ros, the aligned depth should be published under /camera/aligned_depth_to_color/image_raw. 
To get the depth image in rviz, I am showing the topic /camera/depth/image_rect_raw

jolags313, 2023-10-18:
Sorry for the late response, but thank you for the advice- I also had to do a lot of troubleshooting with the camera and firmware to get it to work. I now see the same rviz output you do. However, I am still unable to generate any mesh. Have you figured this out?

shabpompeiano, 2023-10-19:
I suspect based on the following issues, that the issue is related to the TF transformations.

              I had to resolve all the issues related TF and it worked for me.

_Originally posted by @madhu-korada in https://github.com/MIT-SPARK/Kimera-Semantics/issues/75#issuecomment-1483946563_

              Hi @julienip,
              The pose is inferred from the TF tree in ROS (it will look for the tfs you mentioned above, `left_cam` or `left_cam_base_link` depending on what you use).
              Nevertheless, you can also subscribe to a pose topic `transform`, which will be used in case you set the parameter `use_tf_transforms`  to false:
              Note that these are parameters already available in voxblox, and therefore the documentation remains the same, note here https://voxblox.readthedocs.io/en/latest/pages/The-Voxblox-Node.html
              use_tf_transforms
              If true the ros tf tree will be used to get the pose of the sensor relative to the world (sensor_frame and world_frame will be used). If false the pose must be given via the transform topic.

_Originally posted by @ToniRV in https://github.com/MIT-SPARK/Kimera-Semantics/issues/26#issuecomment-615014339_

sguttikon, 2023-11-09:
@shabpompeiano Where you able to solve the issue?